ros-industrial/abb_libegm #7: Protobuf generated headers do not appear to get installed

gavanderhoorn, 2018-04-09:
The headers (and sources) generated by protobuf are not installed by the build script at the moment.

This is less of a problem in a Catkin workpace, as there the include will resolve to the `build` and / or `devel` folder (where the files are stored after code generation), but for a stand-alone build/install (or a user that uses the `install` space) this will not work.

Mitch981521, 2019-03-23:
Has this issue been resolved?  I'm using a catkin workspace and am having issues with the protobuf header files.  The compiler shows the files being compiled, but for some reason they are not able to be found.  Any help would be greatly appreciated.  
[image]
[image]

traversaro, 2019-03-23:
Hi @Mitch981521 , there is a PR that should solve this open at https://github.com/ros-industrial/abb_libegm/pull/23 , but it still needs to be merged in the `master` branch. 
If you want, you can try to apply the same change (that is quite minimal) in your local copy of the repo and see if that fixes your problem.

Mitch981521, 2019-03-23:
@traversaro , is there a certain place in CMakeLists where I should place that?  I just placed it in the installation section where everything else is being installed, but I'm still having the same issue.  A colleague of mine hasn't had any issues with these proto files, so I'm also wondering if there is something wrong with ROS on my computer

gavanderhoorn, 2019-03-24:
@Mitch981521: could you please not post screenshots of terminal text? There is no reason to use images for that, it's all text. Please just copy-paste the text from the console into your comment.

closed 2019-04-11 by gavanderhoorn