ros-industrial/abb_libegm #11: Proto Quaternion definition

YoshuaNava, 2018-06-14:
Hej,
In the [protobuf definition of Quaternion](https://github.com/ros-industrial/abb_libegm/blob/master/proto/egm_wrapper.proto#L115), it is not very clear to which quaternion dimension each field corresponds. e.g.

Is it the case that

```
u1 = x
u2 = y
u3 = z
u4 = w
```

Or

```
u1 = w
u2 = x
u3 = y
u4 = z
```

Thank you.

jontje, 2018-06-14:
Hej!

It is the second one:
```
u1 = w 
u2 = x
u3 = y
u4 = z
```

It follows the same definition as the RAPID data type `orient`

gavanderhoorn, 2018-06-14:
@jontje: would this be something to add as a comment in the `.proto` file?

jontje, 2018-06-14:
Definitely, I will do that

gavanderhoorn, 2019-04-11:
@jontje: not sure, but I believe this one can be closed looking at:

https://github.com/ros-industrial/abb_libegm/blob/a278b7b580f0e2006a9c3c1b1e07dcff9552aef6/proto/egm_wrapper.proto#L126

jontje, 2019-04-11:
@gavanderhoorn: Yes, it can be closed, thanks for the reminder! 👍

closed 2019-04-11 by jontje