ros-industrial/abb_libegm #15: Questions: does abb_libegm allows for joint velocity control?

mitch1044444, 2018-08-04:
Hi All,

I know this perhaps is not the best place to ask questions. However, we are on a tight schedule and have to make a decision recently to purchase a large robot for force control. 

I wonder if abb_libegm allows for the control of joint_velocity in real-time? If so, could you please briefly tell me its real-time performance of joint velocity tracking and its difference from abb_librws?

Thank you a lot for your good contribution for me to ask this question here. Really appreciate your time and help.

jontje, 2018-08-07:
Hi @mitch1044444,

Yes, abb_libegm allows for joint velocity control. Concerning the real-time aspect it will depend on your requirements.

The highest frequency EGM can run at the moment is 250 Hz, so every 4 ms the robot controller can accept new control references. However, there is also a control lag of 10-20 ms depending on the robot type and how EGM has been configured (the numbers are from the official EGM documentation).

Concerning the main difference between abb_libegm and abb_librws:
- abb_libegm: Interface to Externally Guided Motion (EGM), which enables **_streaming_** of motion references. EGM is purely for motion control.
- abb_librws: Interface to Robot Web Services (RWS), which grants access to a number of services and resources in the robot controller. These are interacted with via HTTP requests, and it is not designed for real-time. RWS can be used for motion control, but then it would be more like **_download_** position target(s) to the robot controller and then move.

I hope this helps!

mitch1044444, 2018-08-17:
Thank you so much for your helpful information @jontje . Just want to check if my understanding is correct. 

1. If we have a ABB robot with IRC5 controller, then we install the ABB driver [here](http://wiki.ros.org/abb_driver), abb_libegm and Robotware. I read from ABB websites 

    > RobotStudio includes a matching version of RobotWare. Previous versions can be downloaded from within RobotStudio.

   so we can find a Robotware with the version between 6.0 and 6.06.01. That should allow us to communicate with the robot right?

2. I assume we will need to send EGM commands as String to the abb_libegm library. Also, I wonder if you happen to know where I may find some good examples using EGM? From my own experience, those industry packages usually are not very user-friendly to mechanical engineers... Just want to check if you happen to know them.

You already have helped me a lot. Thank you so much.

mitch1044444, 2018-08-21:
Thank you so much. Really appreciate your help.

closed 2018-08-21 by mitch1044444

Mitch981521, 2019-02-16:
Does anyone know if it is possible to be able to use both a EGMControllerInterface instance and EGMTrajectoryInterface instance in the same program?  We are trying to use EGM to have our robot move to fixed positions but also perform feedback control in the same execution.  The problem we are running into is that it appears that RobotStudio only allows one port for external communication.  Is it possible to somehow use the same port for communication for two different clients?  I've been looking around in the Boost::asio libraries, but I am really unfamiliar in this area.  If someone has any ideas or knows of a library/function that may be able to help, it would be greatly appreciated!

gavanderhoorn, 2019-02-16:
@Mitch981521: please post this as a separate issue. While it's related, your question is actually different from what the OP here (@mitch1044444) asked, which was whether velocity control is supported. Your issue is about support for concurrent use of two classes that are related, but at a different level.

ChunyangXia, 2019-06-02:
> I haven't written down all the details, but this should give you an idea of the requirements.

Hi, Do you know how to set up the instructions in RAPID to  receive  the change of velocity from C++.
Please help me .Thank you very much!

ChunyangXia, 2019-06-03:
> * A RAPID program that executes on the robot controller system. The RAPID program need to initialize the EGM communication, and then activate an EGM communication session (to tell the robot to listen for external motion references).
> 
 @jontje Can you share your PARID code that set up the EMG for changing velocity realtime. This will very helpful. Hope to get your help

jontje, 2019-06-03:
Hello @ChunyangXia,

You should be able to use the RobotWare [StateMachine Add-In](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee).

You can install it via RobotStudio --> Add-Ins tab --> search for StateMachine --> press Add.
Then you can create a new RobotWare system that uses the Add-In. If you do, then the new system will include RAPID code and system configurations.

If you only want the RAPID code, then you can create a dummy virtual system according to the example in the [User Manual](https://robotapps.blob.core.windows.net/appreferences/docs/2093c0e8-d469-4188-bdd2-ca42e27cba5cUserManual.pdf) and extract the code from there.

After you have the RAPID code, then you need to make sure that the RAPID EGMRunX instructions's `PosCorrGain` argument is set to zero (to allow for pure velocity control).

ChunyangXia, 2019-06-03:
@jontje Hi Jontje,
Thanks for your help. Is there any example program or demo for using the "abb_libegm" library. I am not very clear how to use it. Thanks again

gavanderhoorn, 2019-06-03:
Please refer to #18.

ChunyangXia, 2019-06-04:
> After you have the RAPID code, then you need to make sure that the RAPID EGMRunX instructions's `PosCorrGain` argument is set to zero (to allow for pure velocity control).

@jontje 
Thanks for your help. After setting PosCorrGain=0, the velocity can be controled realtime, but the movement lose control, robot will go along X direction until safe guard stop. Do you know why? How to regulate the path of movement under this situation. The following is my RAPID. Thanks again

MODULE EGM_test_UDP
    
    VAR egmident egmID1;
    VAR egmstate egmSt1;
    
    CONST egm_minmax egm_minmax_lin1:=[-1,1]; !in mm
    CONST egm_minmax egm_minmax_rot1:=[-2,2];! in degees
    CONST robtarget p20:=[[34,-499,123],[0.000494947,0.662278052,-0.749217059,-0.007831731],[0,0,-1,0],[9E9,9E9,9E9,9E9,9E9,9E9]];

    PERS tooldata UISpenholder:=[TRUE,[[0,0,114.25],[1,0,0,0]],[1,[-0.095984607,0.082520613,38.69176324],[1,0,0,0],0,0,0]];
    TASK PERS wobjdata wobjBordN:=[FALSE,TRUE,"",[[0,0,0],[1,0,0,0]],[[150,-500,8],[0.707106781,0,0,-0.707106781]]];
    
    VAR pose posecorTable:=[[150,-500,8],[1,0,0,0]];
    VAR pose posesenTable:=[[0,0,0],[1,0,0,0]];

    PROC main()
        ! Move to start position. Fine point is demanded.
        MoveJ p20,v100,fine,UISpenholder;
        testuc_UDP; 
    ENDPROC
  
    
    PROC testuc_UDP()
        EGMReset egmID1;
        EGMGetId egmID1;
        egmSt1 := EGMGetState(egmID1);
        TPWrite "EGM state: "\Num := egmSt1;
        
        IF egmSt1 <= EGM_STATE_CONNECTED THEN
            ! Set up the EGM data source: UdpUc server using device "EGMsensor:"and configuration "default"
            EGMSetupUC ROB_1, egmID1, "default", "EGMSensor:" \pose;
        ENDIF
        
        !Which program to run
        runEGM;
        
        
        IF egmSt1 = EGM_STATE_CONNECTED THEN
            TPWrite "Reset EGM instance egmID1";
            EGMReset egmID1; 
        ENDIF  
    ENDPROC
        
        
        PROC runEGM()
            EGMActPose egmID1\Tool:=UISpenholder \WObj:=wobj0, posecorTable,EGM_FRAME_WOBJ, posesenTable, EGM_FRAME_WOBJ 
            \x:=egm_minmax_lin1 \y:=egm_minmax_lin1 \z:=egm_minmax_lin1
            \rx:=egm_minmax_rot1 \ry:=egm_minmax_rot1 \rz:=egm_minmax_rot1\LpFilter:=2\Samplerate:=4\MaxSpeedDeviation:= 1000;
                
            EGMRunPose egmID1, EGM_STOP_RAMP_DOWN\x\y\z\CondTime:=20 \RampInTime:=0.05\RampOutTime:=0.5\PosCorrGain:=0;
            egmSt1:=EGMGetState(egmID1);
        ENDPROC
 
ENDMODULE

gavanderhoorn, 2019-06-04:
@ChunyangXia: let's not keep commenting on an already closed issue.

Please open a new one.

Steven-ZY, 2023-07-11:
> @jontje Thanks for your help. After setting PosCorrGain=0, the velocity can be controled realtime, but the movement lose control, robot will go along X direction until safe guard stop. Do you know [...]

@ChunyangXia Hi, ChunyangXia. Have you already realized real-time velocity control? May I ask you some questions?