ros-industrial/abb_libegm #17: Expand the readme

jontje, 2018-09-06:
Add sections for:
- Structure/class descriptions
- Troubleshooting
- Acknowledgements

gavanderhoorn, 2018-09-06:
May I suggest we use `Doxygen` and/or `Sphinx` for that?

The code already contains quite a lot of Doxygen compatible comments, and additional material could be added there as well. `Doxygen` has quite good support for this.

gavanderhoorn, 2018-09-06:
re: structure/class descriptions: that's exactly what I would put in the Doxygen docs (see the `roscpp` example I linked earlier).

Those can then be pushed to a gh-pages branch and we'll have them hosted right here on github.

jontje, 2018-09-06:
Ok, great :)

jontje, 2018-09-10:
Follow-up thought:
It would probably not hurt to have the description in both the readme (short version?) and in the API documentation (more detailed version?).

Also, I will probably add some hints on how to build the library on Windows as well.

traversaro, 2018-09-15:
Besides https://github.com/ros-industrial/abb_libegm/issues/3, a few additional fixes will be necessary to correctly compiled the library on  Windows. One issue for sure is symbol visibility, but that can be easily addressed by setting `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` to `TRUE` (see https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/). 

If the library is supposed to be distributed in binary form on Windows, some additonal work may be necessary to create a `relocatable` CMake package (see https://cmake.org/cmake/help/v3.12/manual/cmake-packages.7.html#creating-relocatable-packages), to make sure that the library can be used also if it installed in a location different from the build-time installation destination.

gavanderhoorn, 2019-04-11:
@traversaro: tracking potential Windows blockers in #44.

gavanderhoorn, 2019-04-11:
@jontje: "expand the readme" is essentially an unreachable goal :) We can always expand it further.

Would it make sense to close this issue and open new ones if/when additional updates/extensions to the readme are needed?

jontje, 2019-04-11:
@gavanderhoorn: Yes, I fully agree, I will close this.

closed 2019-04-11 by jontje