ros-industrial/abb_libegm #18: Create code samples

jontje, 2018-09-06:
Create samples to showcase how to use the interface classes, including:
- C++ code
- RAPID code (and possibly a RobotStudio Pack&Go file for a simulation setup)
- Instructions

indraneelpatil, 2019-01-22:
Thank you for raising this issue! If you have a couple of minutes please help me through the situation that I am trying to solve. Please excuse the elementary questions. Thanks in advance.

I am currently trying to send Moveit Joint Trajectories to the 6 axis ABB 1410 manipulator interfaced with the IRC5 controller running Robotware 6.06.01. I successfully used the abb_driver package to implement a ROS socket connection and send these trajectories to the robot but I encountered a latency which I cannot afford for my application.

I am currently trying to set up Externally Guided Motion instead of the abb_driver to get more real time response from the manipulator. I will of course do a sanity check and stream smooth position references through EGM.

As per your instructions I have read the IRC5 Application manual and it has raised the following questions:

1. Are their any server and logger files which must be downloaded to the IRC5 controller like in the case of the abb_driver? If yes then where can I get access to them?
2. As a test setup for using EGM I wish to set up the mouse as a sensor device, would you be able to guide me a little on that?
3. Where does the abb_libegm fit into this entire equation?

Thanks again!

jontje, 2019-01-25:
@indraneelpatil Here are the examples I am working on for the library: [abb_libegm_samples.zip](https://github.com/ros-industrial/abb_libegm/files/2796444/abb_libegm_samples.zip)

Please note that they are not completely finished/fully explained, so there might be some things missing but the code should at least give you some hints about how to use the library. There is also a RobotStudio Pack&Go file included that can be used for simulating the examples in RobotStudio, as well as example RAPID code and system configurations.

I hope this helps!

indraneelpatil, 2019-01-29:
@jontje Your examples were tremendously helpful. Using your examples I could successfully test the following:

1.  I could generate trajectories in moveit and track them in RobotStudio using the EGM joint trajectory mode. The only thing I could not set up was the joint_state_publisher (/joint_states topic) based on feedback from EGM for syncing actual movement with moveit.
2.  I could also use the EGM joint velocity and pose velocity controller node successfully. The only problem I encountered was that in both modes I could not reach the Maximum tool centrepoint velocity of 2.1m/s without getting the 'Dynamic Load Too High Error' on the simulated controller in RobotStudio.

Thank you once again, your help has saved me weeks of work. I am really grateful.

jontje, 2019-01-30:
@indraneelpatil No problem, and great to see that the examples were helpful!

1. You should be able to use the `EGMTrajectoryInterface::retrieveExecutionProgress(...)` method, and then map the retrieved data to the `/joint_states` topic. See the [egm_wrapper_trajectory.proto](https://github.com/ros-industrial/abb_libegm/blob/master/proto/egm_wrapper_trajectory.proto) file for the definition of the `ExecutionProgress` data type.

2. EGM is a bit susceptible to that supervision, which is there to protect the mechanical structure in real robots. You can try to mitigate it by for example:
   * Generate EGM references with less acceleration demands.
   * Tune the parameter `\MaxSpeedDeviation` in the `EGMAct` instructions, in the RAPID code.

indraneelpatil, 2019-01-30:
@jontje Thank you so much for your pointer number 1 regarding the retrieveExecutionProgress method!
Dont know how I missed it. I successfully synced moveit and controller feedback using the following snippet of code :

`ros::Publisher states_pub = node_handle.advertise<sensor_msgs::JointState>("/joint_states", 1000);
  

  
  sensor_msgs::JointState msg;
  msg.header.seq=0;
  msg.name.push_back("joint_1");
  msg.name.push_back("joint_2");
  msg.name.push_back("joint_3");
  msg.name.push_back("joint_4");
  msg.name.push_back("joint_5");
  msg.name.push_back("joint_6");

  for(int k=0;k<6;k++)
  {
    msg.position.push_back(0);
    msg.velocity.push_back(0);
    msg.effort.push_back(0);
  }

  while(ros::ok())
  { 

      egm_interface.retrieveExecutionProgress(&progress);
      if(progress.goal_active())
      {
      msg.header.seq++;
      msg.header.stamp=ros::Time::now();

      for(int m=0;m<6;m++)
        {

         msg.position[m]=0.0174533*progress.inputs().feedback().robot().joints().position().values(m);
         msg.velocity[m]=progress.inputs().feedback().robot().joints().velocity().values(m);

        }

      states_pub.publish(msg);
    }
      ros::spinOnce();
  }`

Thanks again for all the help. On to the next task now!

jontje, 2019-01-30:
@indraneelpatil You are welcome, and good luck with the next task 😃

Mitch981521, 2019-02-04:
I was wondering if someone would be able to assist me with the implementation of the code @jontje provided.  I have robot studio running on a windows computer and have ROS/EGM library on a linux computer that I'm wanting to use to control the robots.  When I run the RAPID code in RobotStudio, I get the following error: "41822 - No data from the UdpUc device".  Here are the steps I've taken so far to try to solve the problem:

-pinged the IP address from linux computer to windows computer and vice versa
-Verified that EGM library option is checked in robot studio
-Changed the IP address in the communication configuration to match the IP address of linux computer
-Increased the "\CommTimeout" argument in the EGMSetupUC procedure from 1 sec to 10 sec

I realize this may go better in the ABB forum, but I'm having issues trying to sign into my account.  Any help would be greatly appreciated!

jontje, 2019-02-05:
@Mitch981521 Just to verify, have you started the EGM UDP server on the linux side? And have you verified that the correct remote port is used in the communication configurations on the RobotStudio side?

Also, have you tried to temporarily turn off the Windows Firewall? It might block the UDP communication, and if it does then you can add rules for the specific ports you want to use.

Mitch981521, 2019-02-05:
I just figured it out....I wasn't changing the IP address for the UCdevice in RobotStudio (I just assumed that only the ROB_# addresses needed changing).  I will make note of the firewall being a potential source of issues.

rsaltus, 2019-03-14:
I'm also having issues with implementing the sample code. Our setup is a Windows machine running RobotStudio connected via ethernet to a router, and a Linux Ubuntu machine running Ubuntu 14.04 and ROS Indigo as our EGM server. The router is connected to the internet via Ethernet. A brief explanation of the procedure undertaken:

Setup Procedure:
1. Imported the Pack&Go file into RobotStudio using RobotWare 6.08.
2. Changed the IP addresses in the Transmission Protocol section of the Communication Configuration tab on the controller so that the IP addresses matched the IP of the Ubuntu EGM Server.
3. Installed and compiled the abb_libegm_samples folder in the ROS workspace. There was a compilation error where catkin couldn't find the protobuf headers, but manually adding the build directory to the CMakeLists.txt file solved this. 
4. Start a ROS Master with the command "roscore", and then open another terminal and run the command "rosrun abb_libegm_samples a1_joint_trajectory_node" to start the UDP server
5. Deselect all tasks other than ROB_1 on RobotStudio, set PP to main, and then start the code.

**An error keeps popping up : "41824 - Not possible to open the UdpUc device", immediately followed by a "40233 - Execution Error".** 

Debugging Steps:
1. Verified the computers can ping each other, so connection is there.
2. Turned off the firewall on the Windows machine to prevent an blocking on the Windows side.
3. Verified the system is on the same subnet, so the UDP connection will not be blocked by any external protocol.
4. Ran command "lsof -i" on Ubuntu side, and verified ROS node is running with port 6511 open.

I have a few ideas about what could possibly be causing the issue, but am not certain. When importing the Pack&Go file, I noticed that it was originally written with RobotStudio version 6.06.01, but abb_libegm now works with strictly versions greater than 6.07.01, so I've imported it using RS 6.08. Are the samples still usable given the updates to the ABB EGM protocol and the updates to the abb_libegm library? If anyone has any advice I'd greatly appreciate it.

danielj195, 2019-03-15:
It looks like you've covered most of the things I would've guessed.  We are using Ubuntu 16.04 and ROS kinetic and were able to get ours to work.  I would disable firewall features on both computers, just for good measure.  In what directory are your compiled protobuf files?  I had to manually compile mine as well and placed them in the include directory, which is where I think they need to go

jontje, 2019-03-15:
@rsaltus, can you retry with this [EGMSamplesSolution Pack&Go.zip](https://github.com/ros-industrial/abb_libegm/files/2970819/EGMSamplesSolution.Pack.Go.zip) file instead?

This Pack&Go is based on RobotWare 6.08, and I can run the samples from the zip-file above with the current version of abb_libegm.

Also, at which RAPID instruction does the error appear?

> An error keeps popping up : "41824 - Not possible to open the UdpUc device", immediately followed by a "40233 - Execution Error".

I hope this helps!

rsaltus, 2019-03-15:
Thanks for your speedy replies, @danielj195 and @jontje. The original error was appearing at the "EGMSetupUC" line. However, I tried the new Pack&Go file this morning and it connects fine, and works as intended. So I think the old RobotWare Pack&Go was causing the problem.

Thank you both for your help.

geohn, 2019-04-30:
> Create samples to showcase how to use the interface classes, including:
> 
> * C++ code
> * RAPID code (and possibly a RobotStudio Pack&Go file for a simulation setup)
> * Instructions

> Create samples to showcase how to use the interface classes, including:
> 
> * C++ code
> * RAPID code (and possibly a RobotStudio Pack&Go file for a simulation setup)
> * Instructions

geohn, 2019-04-30:
@jontje thanks for your help , i want to implement the interpolator method of this C++ code on matlab , but i dont know what's actual defferents between  the two viriables  ‘sample_time’ and 'condition_duration' 
Thanks sincerely!

jontje, 2019-04-30:
@geohn, your question is unrelated to the topic of this issue, so I have moved it to issue https://github.com/ros-industrial/abb_libegm/issues/47.

In the future, please open a new issue instead 😉

shobhitcito, 2020-01-13:
I'm having some issues using the abb_libegm_samples provided by @jontje. I created the linux environment with protobuf & abb_libegm and can build my catkin workspace. But the catkin workspace gives error when I try to build it with abb_libegm_samples.

[error]
c++: fatal error: no input files
compilation terminated.
/bin/sh: 1: -pthread: not found
abb_libegm_samples/CMakeFiles/abb_libegm_samples_joint_trajectory_node.dir/build.make:62: recipe for target 'abb_libegm_samples/CMakeFiles/abb_libegm_samples_joint_trajectory_node.dir/src/a1_joint_trajectory_node.cpp.o' failed
[/error]

Initially there was some error that stated a requirement for c++ 2011 compiler that I resolved by adding "SET(CMAKE_CXX_FLAGS "-std=c++11 -O3") to CMakeLists.txt of abb_libegm_samples & adding the build directory in include_directories. This worked for building with abb_libegm but with abb_libegm_samples, I land up with the above errors.

I am using ubuntu 16.04 & ROS kinetic.

If anybody faced a similar problem, I'd appreciate your advice. Thanks!

shobhitcito, 2020-01-13:
> I'm having some issues using the abb_libegm_samples provided by @jontje. I created the linux environment with protobuf & abb_libegm and can build my catkin workspace. But the catkin workspace gives [...]

My problem got solved, thanks to the suggestion from @traversaro. I removed the line 

list(APPEND CMAKE_CXX_FLAGS -pthread) from /abb_libegm/cmake/abb_libegm-extras.cmake.in

rs-s, 2020-09-11:
Hi, I am unable to build the samples package on Windows ROS Kinetic. Any idea what could be the issue? See below:

C:\catkin_ws>catkin_make_isolated --install --pkg abb_libegm_samples
Base path: C:\catkin_ws
Source space: C:\catkin_ws\src
Build space: C:\catkin_ws\build_isolated
Devel space: C:\catkin_ws\devel_isolated
Install space: C:\catkin_ws\install_isolated
Additional CMake Arguments: -DCMAKE_BUILD_TYPE=RelWithDebInfo
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - abb_libegm (plain cmake)
~~  - abb_libegm_samples
~~  - abb_librws (plain cmake)
~~  - robtest
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Skipping package: 'abb_libegm'

==> Processing catkin package: 'abb_libegm_samples'
==> Building with env: 'C:\catkin_ws\install_isolated\env.bat'
==> cmake C:\catkin_ws\src\abb_libegm_samples -DCATKIN_DEVEL_PREFIX=C:\catkin_ws\devel_isolated\abb_libegm_samples -DCMAKE_INSTALL_PREFIX=C:\catkin_ws\install_isolated -DCMAKE_BUILD_TYPE=RelWithDebInfo -G NMake Makefiles in 'C:\catkin_ws\build_isolated\abb_libegm_samples'
-- Using CATKIN_DEVEL_PREFIX: C:/catkin_ws/devel_isolated/abb_libegm_samples
-- Using CMAKE_PREFIX_PATH: C:/catkin_ws/install_isolated;C:/catkin_ws/devel;C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
-- This workspace overlays: C:/catkin_ws/devel;C:/opt/ros/melodic/x64
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/catkin_ws/build_isolated/abb_libegm_samples/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.28
-- BUILD_SHARED_LIBS is on
-- Could NOT find abb_libegm (missing: abb_libegm_DIR)
-- Could not find the required component 'abb_libegm'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at C:/opt/ros/melodic/x64/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "abb_libegm" with
  any of the following names:

    abb_libegmConfig.cmake
    abb_libegm-config.cmake

  Add the installation prefix of "abb_libegm" to CMAKE_PREFIX_PATH or set
  "abb_libegm_DIR" to a directory containing one of the above files.  If
  "abb_libegm" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build_isolated/abb_libegm_samples/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build_isolated/abb_libegm_samples/CMakeFiles/CMakeError.log".
<== Failed to process package 'abb_libegm_samples':
  Command '['C:\\catkin_ws\\install_isolated\\env.bat', 'cmake', 'C:\\catkin_ws\\src\\abb_libegm_samples', '-DCATKIN_DEVEL_PREFIX=C:\\catkin_ws\\devel_isolated\\abb_libegm_samples', '-DCMAKE_INSTALL_PREFIX=C:\\catkin_ws\\install_isolated', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'NMake Makefiles']' returned non-zero exit status 1
Reproduce this error by running:
==> cd 'C:\catkin_ws\build_isolated\abb_libegm_samples' && 'C:\catkin_ws\install_isolated\env.bat' cmake 'C:\catkin_ws\src\abb_libegm_samples' '-DCATKIN_DEVEL_PREFIX=C:\catkin_ws\devel_isolated\abb_libegm_samples' '-DCMAKE_INSTALL_PREFIX=C:\catkin_ws\install_isolated' -DCMAKE_BUILD_TYPE=RelWithDebInfo -G 'NMake Makefiles'
Command failed, exiting.

gavanderhoorn, 2020-09-11:
@rs-s wrote:
> Hi, I am unable to build the samples package on Windows ROS Kinetic. Any idea what could be the issue?

yes, this:

>     Skipping package: 'abb_libegm'

rs-s, 2020-09-12:
> 
> 
> @rs-s wrote:
> 
> > Hi, I am unable to build the samples package on Windows ROS Kinetic. Any idea what could be the issue?
> 
> yes, this:
> 
> > ```
> > Skipping package: 'abb_libegm'
> > ```

Thanks. I already have abb_libegm installed. I also sourced my windows ROS setup.bat and can see abb_libegm when I do "rospack list" but for some reason it can't be found during the catkin_make isolate process.

It is also looking for abb_libegmConfig.cmake or abb_libegm-config.cmake however abb_libegm only seems to have the following file: abb_libegmConfig.cmake.in

Any ideas?

jontje, 2020-09-14:
Hi @rs-s,

Are you using the plain CMake implementation of `abb_libegm`? If so, have you updated the `abb_libegm_samples`'s `CMakeList.txt` file? I.e. so it doesn't try to find `abb_libegm` as a component of `catkin`.

Note: The `abb_libegm_samples.zip` above was made when `abb_libegm` was a `catkin` package.

> It is also looking for abb_libegmConfig.cmake or abb_libegm-config.cmake however abb_libegm only seems to have the following file: abb_libegmConfig.cmake.in

`abb_libegmConfig.cmake.in` is used to generate the `abb_libegmConfig.cmake` file when you run `catkin_make_isolated`. Have you checked your workspace's build/devel/install folders to see if `abb_libegm` was built and installed? `abb_libegmConfig.cmake` should be in there if it was.

cpilon, 2020-12-07:
> Hi @rs-s,
>
> Are you using the plain CMake implementation of `abb_libegm`? If so, have you updated the `abb_libegm_samples`'s `CMakeList.txt` file? I.e. so it doesn't try to find `abb_libegm` as a [...]

Hi @jontje ,

I am experiencing similar issues trying to build your samples with the plain CMake implementation. Could you please provide more details on what modifications need to be made to the abb_libegm_samples's CMakeList.txt? 

I have removed links to abb_libegm from the CMakeList.txt and modified the include statements in the cpp files to point to the abb_libegm headers but now I get the following error **"fatal error: abb_libegm_export.h: No such file or directory"**. However I have built abb_libegm successfully.

Any help would be much appreciated.

jontje, 2020-12-07:
Hi @cpilon,

> I am experiencing similar issues trying to build your samples with the plain CMake implementation. Could you please provide more details on what modifications need to be made to the abb_libegm_samples's CMakeList.txt?

What command(s) do you run when building?

> I have removed links to abb_libegm from the CMakeList.txt and modified the include statements in the cpp files to point to the abb_libegm headers but now I get the following error **"fatal error: abb_libegm_export.h: No such file or directory"**. However I have built abb_libegm successfully.

You should have something like this in the`CMakeList.txt` file:
- `find_package(abb_libegm REQUIRED)`
- `target_link_libraries(... abb_libegm::abb_libegm ...)`

cpilon, 2020-12-07:
Hi @jontje  - thanks for your reply.

> What command(s) do you run when building?
I use catkin_make_isolated. 
abb_libegm and abb_librws were built successfully using this. 

> You should have something like this in the`CMakeList.txt` file:
> 
>     * `find_package(abb_libegm REQUIRED)`
Ok, I added this.

>     * `target_link_libraries(... abb_libegm::abb_libegm ...)`
This seems to be my issue. Where could I find more info about the right way to format this part?

I have made a change in my include statement in the samples cpp files. For now I have hardcoded my path to #include "my_workspace/devel_isolated/abb_libegm/include/abb_libegm/egm_trajectory_interface.h". Now it can at least find the abb_libegm_export.h I mentioned earlier. However, I got the following errors because I guess I have not defined the abb and egm namespaces properly:

`a1_joint_trajectory_node.cpp:(.text+0x191): undefined reference to `abb::egm::EGMTrajectoryInterface::EGMTrajectoryInterface(boost::asio::io_service&, unsigned short, abb::egm::TrajectoryConfiguration const&)'
a1_joint_trajectory_node.cpp:(.text+0x1a0): undefined reference to `abb::egm::EGMBaseInterface::isInitialized()'
a1_joint_trajectory_node.cpp:(.text+0x468): undefined reference to `abb::egm::wrapper::trajectory::TrajectoryGoal::TrajectoryGoal()'
a1_joint_trajectory_node.cpp:(.text+0x477): undefined reference to `abb::egm::wrapper::trajectory::TrajectoryGoal::TrajectoryGoal()'
a1_joint_trajectory_node.cpp:(.text+0xedb): undefined reference to `abb::egm::EGMBaseInterface::isConnected()'
a1_joint_trajectory_node.cpp:(.text+0xefc): undefined reference to `abb::egm::EGMBaseInterface::getStatus()'
a1_joint_trajectory_node.cpp:(.text+0xf1f): undefined reference to `abb::egm::wrapper::Status::~Status()'
a1_joint_trajectory_node.cpp:(.text+0x1064): undefined reference to `abb::egm::EGMBaseInterface::getStatus()'
a1_joint_trajectory_node.cpp:(.text+0x108e): undefined reference to `abb::egm::wrapper::Status::~Status()'
a1_joint_trajectory_node.cpp:(.text+0x120d): undefined reference to `abb::egm::wrapper::trajectory::TrajectoryGoal::TrajectoryGoal(abb::egm::wrapper::trajectory::TrajectoryGoal const&)'`

jontje, 2020-12-08:
> This seems to be my issue. Where could I find more info about the right way to format this part?

You could for example read a tutorial on modern `CMake`,

> I have made a change in my include statement in the samples cpp files. For now I have hardcoded my path to #include [...]

You should really try to avoid hardcoded include paths, it is not good for portability.

Here is a rough example package ([abb_egm_sample.zip](https://github.com/ros-industrial/abb_libegm/files/5660559/abb_egm_sample.zip)) for linking `abb_libegm` into a `ROS` node.

It works for me on `Ubuntu` `18.04` with `ROS` `Melodic` and the latest version of `abb_libegm`.

I built it with `catkin_make_isolated`.

I hope this helps!

cpilon, 2021-01-11:
@jontje - thanks for your help, I have been able to figure out my issues!

ScottYang1982, 2021-02-13:
@jontje 
The latest abb_libegm, using **catkin build** on ubuntu 16.04, ROS kinetic.
abb_libegm built ok but when tried to **catkin build** abb_egm_sample, it said cant find package **abb_libegm**.

ScottYang1982, 2021-02-13:
Errors     << abb_egm_sample:cmake /root/egm_samples_ws/logs/abb_egm_sample/build.cmake.001.log                                                                                                                                                                                                               
CMake Error at /root/egm_samples_ws/src/abb_egm_sample/CMakeLists.txt:11 (find_package):
  By not providing "Findabb_libegm.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "abb_libegm", but CMake did not find one.

  Could not find a package configuration file provided by "abb_libegm" with
  any of the following names:

    abb_libegmConfig.cmake
    abb_libegm-config.cmake

  Add the installation prefix of "abb_libegm" to CMAKE_PREFIX_PATH or set
  "abb_libegm_DIR" to a directory containing one of the above files.  If
  "abb_libegm" provides a separate development package or SDK, be sure it has
  been installed.

jontje, 2021-02-15:
> @jontje
> The latest abb_libegm, using **catkin build** on ubuntu 16.04, ROS kinetic.
> abb_libegm built ok but when tried to **catkin build** abb_egm_sample, it said cant find package **abb_libegm**.

How does your ROS catkin workspace(s) structure look like?

Did you build abb_libegm in an underlay workspace? If so, have you "chained it" with an overlay workspace? Or are you building both the abb_libegm and abb_egm_sample packages in the same workspace?

ScottYang1982, 2021-02-17:
Hi @jontje ,
abb_libegm and samples are built separately. Using findpackage when building samples.
How to link them together correctly?

gavanderhoorn, 2021-02-17:
@ScottYang1982: please show us the exact steps you're using to create your workspace, place the packages in it, install the dependencies and build it.

As @jontje wrote: there are a few steps there which will directly influence the success of what you're trying to do.

Without more information on what you're already doing, we cannot help you.

stevedanomodolor, 2021-03-04:
I was able to compile the codes and was able to run the code but I get stuck in the wait for an EGM communication to start from thr ros side. 

When I start the simulation from the robotstudio side, i get 
```
EGMSamplesSystem (Station): 10149 - Program Pointer moved to routine [4]	04/03/2021 08:00:15	Event Log	
EGMSamplesSystem (Station): 10052 - Regain start	04/03/2021 08:00:15	Event Log	
EGMSamplesSystem (Station): 10053 - Regain ready	04/03/2021 08:00:15	Event Log	
EGMSamplesSystem (Station): 10151 - Program started [4]	04/03/2021 08:00:15	Event Log	
EGMSamplesSystem (Station): 41822 - No data from the UdpUc device [4]	04/03/2021 08:00:17	Event Log	
EGMSamplesSystem (Station): 50501 - Short movements [4]	04/03/2021 08:00:22	Event Log	
EGMSamplesSystem (Station): 41822 - No data from the UdpUc device [4]	04/03/2021 08:00:23	Event Log	
EGMSamplesSystem (Station): 10125 - Program stopped [4]	04/03/2021 08:00:27	Event Log	

```

ROs side 

```
[ INFO] [1614841087.905672514]: ========== Joint trajectory sample ==========
[ INFO] [1614841087.906224081]: 1: Wait for an EGM communication session to start...

```

Am I  supposed to do something else to connect the two o just change the port address in the ros code?, 
Also, How do I deselect all tasks other than ROB_1 on RobotStudio, set PP to main?

Thank you very much in advansce 
Note: 
I am using ubuntu 20 running on a virtual machine.  and running robotsudio on windows

gavanderhoorn, 2021-03-04:
> Am I supposed to do something else to connect the two o just change the port address in the ros code?,

Did you setup the `UdpUc` device? You need to make sure you've configured the correct IP address for it to be able to communicate with the UDP server (ie: the code you run yourself).

You can find it via ` RobotStudio > Controller tab > Configuration > Communication > Transmission Protocol`.

This is also mentioned in the FAQ of the `abb_robot_driver`. See [here](https://github.com/ros-industrial/abb_robot_driver/tree/20d888d4891e62d3734c6be9d0ab759618a9e0c5/abb_egm_hardware_interface#troubleshooting).

jontje, 2021-03-04:
> Also, How do I deselect all tasks other than ROB_1 on RobotStudio, set PP to main?

You can use `Ctrl + Shift + M` to reset all program pointers to main, and select RAPID tasks to run via `RobotStudio > RAPID tab > Selected Tasks > Check/uncheck tasks `.

For example:
[image]

stevedanomodolor, 2021-03-06:
@jontje thank you very much for your answer, i was able to connect and run the programs. 
I have some question that might sound silly. 
- What is the difference between using the egm library or the abb_driver library?

gavanderhoorn, 2021-03-06:
@stevedanomodolor: let's keep this on-topic.

Please post a new issue on the appropriate issue tracker.