ros-industrial/abb_libegm #20: Add support for additional EGM modes

jontje, 2018-09-06:
As per title.

Currently unsupported modes are:
- **EGM Path Correction**
Path correction can be used to correct a programmed robot path (requires using special RAPID move instructions).

- **EGM Position Stream (requires RobotWare 6.07 or later)**
Position streaming can be used for when it is only interesting to track a mechanical unit's position, and not to control it's motion.

This can for example be done by implementing additional EGM interface classes.

gavanderhoorn, 2019-02-21:
@jontje: this should now be possible with the merge of #32 right?

jontje, 2019-02-22:
Yes it is

gavanderhoorn, 2020-02-05:
@jontje: is there still work to be done here?

jontje, 2020-02-06:
I will not be implementing support for `EGM Path Correction` anytime soon.

Regarding `EGM Position Stream` (i.e. only get feedback from the robot, and not send commands), then it works with the current [EGMControllerInterface](https://github.com/ros-industrial/abb_libegm/blob/89791927dd25c99df0344fa6e23279579868bf38/include/abb_libegm/egm_controller_interface.h) class. However, it might be nice to create an `EGMStreamingInterface` that only exposes a read method.

I also intend to implement support for external axes, which at the moment needs to be controlled in its own RAPID motion task (the `abb_libegm` classes currently only work when a six or seven axes robot is "on the other end"). And here it might also be nice to implement an `EGMExternalAxisInterface`, which doesn't expose the Cartesian write methods.