ros-industrial/abb_libegm #30: Support for two different classes

danielj195, 2019-02-16:
I was wondering if it is possible to concurrently use the egmControllerInterface and egmTrajectoryInterface classes in the same program.  Robotstudio only allows one port per robot, and I'm not sure if it's even possible for more than one socket to connect to the same port.  Any help would be greatly appreciated!

jontje, 2019-02-18:
### Answer
Yes, it is possible to use multiple instances of the interface classes in the same program. The only requirement is that each instance binds to a separate port number.

On the robot controller side, you need to add the corresponding `Transmission Protocol` instances in the `System Configuration`. You can do this via RobotStudio:
* Controller tab -> Configuration -> Communication -> Transmission Protocol -> Add new instances (of type UDPUC) for each port number you want to use.

Each mechanical unit in the robot controller can then use any of those `Transmission Protocol` instances when setting up the EGM communication (with the `EGMSetupUC` instruction).

However, only one port can be active for a single mechanical unit at the same time. If you want a robot to switch to another instance, then you need to reset the EGM communication for that robot, and then setup the EGM communication again with a different `Transmission Protocol` instance.

I wouldn't really recommend this approach since it takes some seconds at least to tear down/set up the EGM communication. Especially if there are EGM motions executing.

### Possible alternative
By the way, are you going to use the `EGMControllerInterface` and `EGMTrajectoryInterface` instances with the same robot? If so, have you tried the `EGMTrajectoryInterface`'s static goal execution mode?

Then you wouldn't maybe need the `EGMControllerInterface` instance, since the static goal execution mode allows for overriding the current trajectory execution. Then you can set the static goal to follow. The static goal can be changed online, but it hasn't been designed for frequent updates.

I would recommend this approach, if it is enough for your use case.

I hope this helps!

danielj195, 2019-02-18:
@jontje Thank you for the response!  Do you know if there will be any future versions of this library that will allow for concurrent use of these classes?

gavanderhoorn, 2019-02-18:
(as this is not an issue with the library in this repository I'm closing this to keep the tracker clean and focus on actionable items/actual bugs: do please feel free to keep commenting on it @danielj195).

closed 2019-02-18 by gavanderhoorn

jontje, 2019-02-19:
@danielj195 You are welcome.

> Do you know if there will be any future versions of this library that will allow for concurrent use of these classes?

No, there are no such plans for using two interfaces, at the same time, with a single robot. It wouldn't really make sense due to how EGM communication works. It would be more appropriate to expand one of the existing interfaces (or implement a new type of interface) with the desired features.

Also, as it is now, then you can use two interfaces with two separate robots at the same time.