ros-industrial/abb_libegm #41: Restarting EGM Session properly

BidoMan, 2019-03-16:
Hi,

I am trying to implement a continuously running system that controls the robot on RobotStudio, that can also mitigate automatically from  RobotStudio runtime errors like **Dynamic Load** and **Joint Out Of Range** without me having to manually restart the system or code.

I am currently using two ros nodes:
**The First** one using **abb_librws** with **State Machine Add-in** to restart motion, handle runtime errors, and to send trajecotry data to abb_libegm.
**The Second** is using **abb_libegm** to send the calculated trajectories to the robot.

I found that the best way to handle RobotStudio runtime errors, specially **Dynamic Load** was to close egm ros node and run it again, so is there a proper way to restart egm session automatically without me having to manually shutdown the node?

Example execution sequence that gives **Dynamic Load**:

First use MoveAbsJ to go to starting position
Second execute EGM trajectory to go to desired goal.

Then repeat with the same EGM trajectory.

After executing this cycle several times, then the problem of **Dynamic Load** occurs, and i can't mitigate from it unless i manually intervene, i tried stopping the rapid execution and starting it again from abb_librws node but it just keeps giving the dynamic load every time.

Thanks.

jontje, 2019-03-18:
Hi @BidoMan,

You shouldn't need to restart the ROS nodes.

**Regarding recovery behaviour:**
I am assuming that you are using the [RWSStateMachineInterface](https://github.com/ros-industrial/abb_librws/blob/master/include/abb_librws/rws_state_machine_interface.h) class of abb_librws.

How does your code look like for restarting the RAPID program?

Something like this works fine for me  when I for example want to restart after a **Dynamic Load** error. 
```
std::string rapid_task = abb::rws::SystemConstants::RAPID::TASK_ROB_1;

abb::rws::RWSStateMachineInterface rws_interface("127.0.0.1");

// Try to restart the RAPID program as long as:
// * The RAPID program is not running.
// * The StateMachine RAPID program is not in it's idle state.
while (ros::ok() &&
       (!rws_interface.isRAPIDRunning().isTrue() ||
        !rws_interface.services().main().isStateIdle(rapid_task).isTrue()))
{
  if (!rws_interface.isMotorOn().isTrue())
  {
    rws_interface.setMotorsOn();
  }
  if (rws_interface.isRAPIDRunning().isTrue())
  {
    rws_interface.stopRAPIDExecution();
  }
  rws_interface.resetRAPIDProgramPointer();
  rws_interface.startRAPIDExecution();

  ros::Duration(0.5).sleep();
}
```

I have only made a quick test, but this should work fine as long as the robot controller is in `Auto` mode.

**Regarding abb_libegm:**
Which class are you using from the library? E.g.: [EGMControllerInterface](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_controller_interface.h) or [EGMTrajectoryInterface](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_trajectory_interface.h)?

BidoMan, 2019-03-18:
I'm using **EGMTrajectoryInterface**.

I solved the problem by adding a delay after each motion command by about 1 second and adding a delay of 0.01 seconds after each communication request, it appears that it was a synchronization problem between my **abb_librws node** and the Rapid code using the **StateMachine Add-in**, that caused motion to be executed at the wrong time (giving Dynamic Load), also causing unsuccessful or wrong timed communication requests like setting/resetting I/O signals. 

The communication delay was to prevent **Server Unavailable** communication error that comes when trying to make a lot of requests in a short time and to also keep it synchronized with Rapid code.

The motion delay was to allow for some time for the motion to start such that when i for example want to know if the robot is stationary **rws_interface.Is_Stationary.IsTrue()** it would be due to it finishing the motion or an error and not because it didn't start the motion yet.

jontje, 2019-03-19:
Ok, good that you solved it.

And just to let you know, then PR https://github.com/ros-industrial/abb_libegm/pull/40 will improve the internal management of the **EGMTrajectoryInterface** class. One of the changes makes the interface to discard all pending trajectories when an active EGM session is closed or when an error is detected in the session. This basically results in the same behaviour as your usage of **EGMTrajectoryInterface::stop(true)**, if I understood you correctly.

gavanderhoorn, 2019-09-26:
Closing as the original issue seems to have been dealt with.

We can always re-open if required, and feel free to keep commenting on this issue.

closed 2019-09-26 by gavanderhoorn