ros-industrial/abb_libegm #42: EGM read joint angle always 57 times larger

Thompson104, 2019-03-22:
Hi All,

We were using the EGM to do real-time control and want to move the robot 4 joint, for example, from 0 degree to 30 degree. The simulation on robot studio works just fine. However, the reading becomes 0- 1710, instead of 0 - 30, when we run the experiments on the real robot.

I wonder if you may know what's going on. The customer service agent was not responding to our questions since they don't know this EGM library...

gavanderhoorn, 2019-03-22:
Which version of RobotWare do you have installed on the controller?

And which version of `abb_libegm` are you using?

Thompson104, 2019-03-22:
@gavanderhoorn Thank you for your quick response. We were using the 6.06 robotware and I believe we have the most recent abb_libegm.

gavanderhoorn, 2019-03-22:
> We were using the 6.06 robotware and I believe we have the most recent abb_libegm.

That would be the problem then.

Thompson104, 2019-03-22:
Oh man, thank you. Previously the abb_libegm does not work with robotware versions earlier than the 6.07. I didn't notice the update. 

Thank you.

Thompson104, 2019-03-22:
We will be testing the robot now to see if works.

gavanderhoorn, 2019-03-22:
Closing this as it would seem to be a user configuration issue.

gavanderhoorn, 2019-03-22:
@jontje: would there be some way we could add some sort of version detection to `abb_libegm` to prevent situations like this?

With `abb_librws` that would probably be possible, but with just `abb_libegm`?

closed 2019-03-22 by gavanderhoorn

jontje, 2019-04-11:
> @jontje: would there be some way we could add some sort of version detection to `abb_libegm` to prevent situations like this?
> 
> With `abb_librws` that would probably be possible, but with just `abb_libegm`?

It should be possible to detect if proto fields introduced in the new EGM protocol is missing or not, I will investigate.

gavanderhoorn, 2019-04-18:
Tracking version checking in #46.