ros-industrial/abb_libegm #44: Investigate compatibility of build scripts with Windows

gavanderhoorn, 2019-04-11:
Promoting https://github.com/ros-industrial/abb_libegm/issues/17#issuecomment-421624204 by @traversaro to this issue:
> Besides #3, a few additional fixes will be necessary to correctly compiled the library on Windows. One issue for sure is symbol visibility, but that can be easily addressed by setting `CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS` to `TRUE` (see https://blog.kitware.com/create-dlls-on-windows-without-declspec-using-new-cmake-export-all-feature/).
> 
> If the library is supposed to be distributed in binary form on Windows, some additonal work may be necessary to create a `relocatable` CMake package (see https://cmake.org/cmake/help/v3.12/manual/cmake-packages.7.html#creating-relocatable-packages), to make sure that the library can be used also if it installed in a location different from the build-time installation destination.

traversaro, 2019-04-11:
Related documentation from the ROSOnWindows project: https://ms-iot.github.io/ROSOnWindows/Porting/Cookbook .

gavanderhoorn, 2019-04-11:
`abb_libegm` is not a ROS package necessarily, but it's a nice checklist.

gavanderhoorn, 2020-05-10:
Related: #91.

gavanderhoorn, 2023-02-22:
Not really a build script per se, but Windows CI broke again: #156.

This needs more effort than just updating a single line, so just tracking it here for now.