ros-industrial/abb_libegm #45: Trouble Connecting to Robotstudio

danielj195, 2019-04-17:
I'm having issues getting abb_libegm to connect to both robotstudio and the robot. I've tried several things to try to remedy the issue.  Here is a list of items that I verified before posting here:

-Made sure IP addresses are in same domain
-Compiled .proto files using protobuf compiler (version 2.6.1)
-Performed a clean catkin compile before trying to run code
-Made sure that Configuration IP addresses in robotstudio match IP address of remote computer
-Tried connecting using sample code, including the Pack&Go file
- Restarted both the robotstudio rapid program and the controller
- Disabled firewalls

We've got EGM working before in the past, but it is no longer connecting for some unknown reason.  Any help would be greatly appreciated.

gavanderhoorn, 2019-04-18:
If you receive any specific error messages it would be good to show them.

danielj195, 2019-04-18:
yeah we were able to ping.  The error was "No data from the UdpUc" in robotstudio.  We figured it out though...our robotstudio program was robotware 6.06, which is not compatible with EGM.  Thanks for getting back to me

gavanderhoorn, 2019-04-18:
Ah.

I've just opened #46 to track the issues with older RWs and `abb_libegm`.

Thanks for reporting back.

closed 2019-04-18 by gavanderhoorn