ros-industrial/abb_libegm #46: Investigate version detection of EGM protocol

gavanderhoorn, 2019-04-18:
With the requirement of using at least version `6.07.01` of RobotWare in #32, we've had a few reports of users running into issues when trying to use an older version of RW (see #42 and #45).

It would be good if `abb_libegm` supported some sort of version check, as this could help us print an error message if/when an incompatible version is being used.

gavanderhoorn, 2019-11-26:
With a wider range of RW 6 versions available now, the potential for unexpectedly running into an incompatible RW version has increased.

Have you had any time to look at how to implement some form of version detection @jontje?

jontje, 2019-11-26:
> Have you had any time to look at how to implement some form of version detection @jontje?

Not yet, but I have an idea of how to do it.

I have a couple minor PRs prepared after this one https://github.com/ros-industrial/abb_libegm/pull/63, and after that I am working on another related to `RW6.10`. I can investigate about version checking in that one.

gavanderhoorn, 2020-02-05:
@jontje: friendly ping again :)

We have even more RW6 versions now then when I wrote https://github.com/ros-industrial/abb_libegm/issues/46#issuecomment-558544213, so this may be something valuable to have (ie: the ability to detect versions, and block incompatible ones).

jontje, 2020-02-06:
> @jontje: friendly ping again :)

Thanks for the reminder 😃

PR #81 (2020-02-19): Add version detection
Fix #46 

As a side note, then this also means that I could add support for RW versions less than `6.07` again.

Edit: I forgot to mention that this is only a rough version estimation, based on if some fields exists or not.
closed 2020-02-19 by gavanderhoorn