ros-industrial/abb_libegm #47: Question: Meaning of ‘sample_time’ and 'condition_duration' in EGMInterpolator class?

jontje, 2019-04-30:
Moved question asked by @geohn in issue https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-487833062 to a new issue.

> @jontje thanks for your help , i want to implement the interpolator method of this C++ code on matlab , but i dont know what's actual defferents between the two viriables ‘sample_time’ and 'condition_duration'
> Thanks sincerely!

The [EGMInterpolator](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_interpolator.h) class has the following subclasses for calculating interpolations:
* [SplinePolynomial](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_interpolator.h#L260-L265) 
* [Slerp](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_interpolator.h#L366-L377)
* [SoftRamp](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_interpolator.h#L447-L456)

`SplinePolynomial` and `Slerp` are used when following a known trajectory (with one or more goals that doesn't change), while `SoftRamp` is used when only either a position or a velocity goal (that can be changed) is known.

All of them uses `condition_duration` to specify the time it should take to reach the goal, while `SoftRamp` uses `sample_time` for estimations (i.e. estimate position if velocity goal and vice versa).

I hope this helps!

gavanderhoorn, 2019-08-26:
@jontje: is this something actionable? Or can we close this?

gavanderhoorn, 2019-09-26:
I believe @jontje answered a question posted in https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-487833062 here.

As such, this is not an actionable issue. Closing.

closed 2019-09-26 by gavanderhoorn