ros-industrial/abb_libegm #50: Movement lose control

jontje, 2019-06-10:
New issue for @ChunyangXia's question from https://github.com/ros-industrial/abb_libegm/issues/15#issuecomment-498528184.

I.e.:
> > After you have the RAPID code, then you need to make sure that the RAPID EGMRunX instructions's `PosCorrGain` argument is set to zero (to allow for pure velocity control).
> 
> @jontje
> Thanks for your help. After setting PosCorrGain=0, the velocity can be controled realtime, but the movement lose control, robot will go along X direction until safe guard stop. Do you know why? How to regulate the path of movement under this situation. The following is my RAPID. Thanks again

@ChunyangXia, did you manage to solve this? I have been out of office and just got back today.

I did a quick test with RobotStudio simulations, and since I don't know your setup I just used the following:

- RobotStudio:
  - RobotWare 6.08.01
  - An IRB1600 robot
  - Your RAPID code
- C++ program:
  -  [EGMControllerInterface](https://github.com/ros-industrial/abb_libegm/blob/master/include/abb_libegm/egm_controller_interface.h)
  - Generated a mock sinusoidal velocity on the x-axis

The robot moved as expected for me.

If this is still an issue, could you provide some details about your actual setup? And what did you mean with "movement lose control"? E.g. did the robot stop responding to your EGM references?

ChunyangXia, 2019-06-11:
> And what did you mean with "movement lose control"? E.g. did the robot stop responding to your EGM references?

@jontje Thank you. The robot don't move along the trajectory in program. It move along y axis util security guard stop. The program I used is [EGM slider](https://github.com/orjanmehre/EGM-Slider), and I add velocity control code in it, and set PosCorrGain=0 in rapid:
            EgmSpeedRef.Builder s1 = new EgmSpeedRef.Builder();   
            EgmCartesianSpeed.Builder s2 = new EgmCartesianSpeed.Builder();
            s2.AddValue(WeldSpeed);     
            s1.SetCartesians(s2);         
            sensor.SetSpeedRef(s1);
The velocity could vary realtime, but the move trajectory can not be controlled.

jontje, 2019-06-11:
@ChunyangXia, aha, I see.

The problem is that you have set that the EGM references should be relative to the `wobj0` work object.

I assume that you actually want to use the work object `wobjBordN` that you have defined in the top of the RAPID module, which is rotated -90 degrees around the z-axis compared to `wobj0`.

You need to replace this:

`EGMActPose egmID1\Tool:=UISpenholder \WObj:=wobj0, posecorTable,EGM_FRAME_WOBJ, posesenTable, EGM_FRAME_WOBJ 
        \x:=egm_minmax_lin1 \y:=egm_minmax_lin1 \z:=egm_minmax_lin1
        \rx:=egm_minmax_rot1 \ry:=egm_minmax_rot1 \rz:=egm_minmax_rot1\LpFilter:=2\Samplerate:=4\MaxSpeedDeviation:= 1000;`

With this (notice the `\WObj` argument):

`EGMActPose egmID1\Tool:=UISpenholder \WObj:=wobjBordN, posecorTable,EGM_FRAME_WOBJ, posesenTable, EGM_FRAME_WOBJ 
        \x:=egm_minmax_lin1 \y:=egm_minmax_lin1 \z:=egm_minmax_lin1
        \rx:=egm_minmax_rot1 \ry:=egm_minmax_rot1 \rz:=egm_minmax_rot1\LpFilter:=2\Samplerate:=4\MaxSpeedDeviation:= 1000;`

gavanderhoorn, 2019-08-26:
Closing due to inactivity.

closed 2019-08-26 by gavanderhoorn