ros-industrial/abb_libegm #55: can't build abb_libegm_samples

ejbkdb, 2019-08-26:
I'm having difficulty building the abb_libegm_samples from: [issue 18](https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-457592204)

I downloaded a protoc executable from: [link](https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1), copied it to /usr/bin and chmod +x protoc so it would run, I was able to build abb_libegm. I was not able to experience the same success when I tried to build the samples file.

which protoc = /usr/bin/protoc
protoc version = libprotoc 3.6.1

See output below.

Any ideas on how to resolve: undefined reference to `google::protobuf::xxxxx?

> CMakeFiles/abb_libegm_samples_joint_trajectory_node.dir/src/a1_joint_trajectory_node.cpp.o: In function `google::protobuf::Arena::AllocHook(std::type_info const*, unsigned long) const':
a1_joint_trajectory_node.cpp:(.text._ZNK6google8protobuf5Arena9AllocHookEPKSt9type_infom[_ZNK6google8protobuf5Arena9AllocHookEPKSt9type_infom]+0x3d): undefined reference to `google::protobuf::Arena::OnArenaAllocation(std::type_info const*, unsigned long) const'
CMakeFiles/abb_libegm_samples_joint_trajectory_node.dir/src/a1_joint_trajectory_node.cpp.o: In function `google::protobuf::RepeatedField<double>::Reserve(int)':
a1_joint_trajectory_node.cpp:(.text._ZN6google8protobuf13RepeatedFieldIdE7ReserveEi[_ZN6google8protobuf13RepeatedFieldIdE7ReserveEi]+0x359): undefined reference to `google::protobuf::internal::ArenaImpl::AllocateAligned(unsigned long)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::ArenaImpl::AllocateAlignedAndAddCleanup(unsigned long, void (*)(void*))'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::DecrementRecursionDepthAndPopLimit(int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::AssignDescriptors(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadVarint64Fallback()'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::UnknownFieldSet::default_instance()'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::IncrementRecursionDepthAndPushLimit(int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::DestroyMessage(void const*)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadVarint32Fallback(unsigned int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::InitSCCImpl(google::protobuf::internal::SCCInfoBase*)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::Message::SpaceUsedLong() const'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::SkipFallback(int, int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm//libhome//libabb_libegm.soejbkdb:/ sandingundefined/ catkin_wsreference/ develto/ .`privategoogle/:abb_libegm:/protobuflib:/:libabb_libegm.soio:: :undefinedCodedInputStream :reference: ReadVarintSizeAsIntFallbackto( )`'google
:/:homeprotobuf/:ejbkdb:/internalsanding:/:catkin_wsArenaImpl/:devel:/AllocateAlignedAndAddCleanup.(privateunsigned/ abb_libegmlong/,lib /voidlibabb_libegm.so :( *undefined) (referencevoid *to) )`'google
:/:homeprotobuf/:ejbkdb:/iosanding:/:catkin_wsCodedInputStream/:devel:/ReadTagFallback.(privateunsigned/ abb_libegmint/)lib'/
libabb_libegm.so/:home /undefinedejbkdb /referencesanding /tocatkin_ws /`develgoogle/:.:privateprotobuf/:abb_libegm:/iolib:/:libabb_libegm.soCodedInputStream:: :undefinedDecrementRecursionDepthAndPopLimit (referenceint )to' 
`/googlehome:/:ejbkdbprotobuf/:sanding:/internalcatkin_ws:/:develOnShutdownRun/(.voidprivate /(abb_libegm*/)lib(/voidlibabb_libegm.so :const *undefined) ,reference  voidto  const`*google):':
protobuf/:home:/internalejbkdb:/:sandingAssignDescriptors/(catkin_wsstd/:devel:/__cxx11.:private:/basic_stringabb_libegm</charlib,/ libabb_libegm.sostd:: :undefinedchar_traits <referencechar >to,  `stdgoogle::::allocatorprotobuf<:char:>internal :>: RepeatedPtrFieldBaseconst:&:,InternalExtend (googleint:):'protobuf
::internal::MigrationSchema const*, google::protobuf::Message collect2: error: ld returned 1 exit status
const* const*, unsigned int const*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadVarint64Fallback()'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::UnknownFieldSet::default_instance()'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::IncrementRecursionDepthAndPushLimit(int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::DestroyMessage(void const*)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadVarint32Fallback(unsigned int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::InitSCCImpl(google::protobuf::internal::SCCInfoBase*)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::Message::SpaceUsedLong() const'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::SkipFallback(int, int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::Arena::OnArenaAllocation(std::type_info const*, unsigned long) const'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::ArenaImpl::AllocateAligned(unsigned long)'
/make[2]: *** [/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm_samples/lib/abb_libegm_samples/a1_joint_trajectory_node] Error 1
home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadVarintSizeAsIntFallback()'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::io::CodedInputStream::ReadTagFallback(unsigned int)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/make[1]: *** [CMakeFiles/abb_libegm_samples_joint_trajectory_node.dir/all] Error 2
libabb_libegm.so: undefined reference to `google::protobufmake[1]: *** Waiting for unfinished jobs....
::internal::OnShutdownRun(void (*)(void const*), void const*)'
/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm/lib/libabb_libegm.so: undefined reference to `google::protobuf::internal::RepeatedPtrFieldBase::InternalExtend(int)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ejbkdb/sanding/catkin_ws/devel/.private/abb_libegm_samples/lib/abb_libegm_samples/a2_pose_trajectory_node] Error 1
make[1]: *** [CMakeFiles/abb_libegm_samples_pose_trajectory_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/ejbkdb/sanding/catkin_ws/build/abb_libegm_samples; catkin build --get-env abb_libegm_samples | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

ejbkdb, 2019-08-26:
I reinstalled ubuntu, ros, ros-industrial, abb stack, abb_libegm, and protoc, and was successfully able to build the library. Maybe I had a protoc conflict???

closed 2019-08-26 by ejbkdb

jontje, 2019-08-26:
Great that you solved the issue.

> Maybe I had a protoc conflict???

That was my initial thought when I read your issue. It's best to have the same version of `libprotobuf` and the `protoc` compiler.

gavanderhoorn, 2019-08-26:
If you're using Ubuntu, I'd recommend installing the `protobuf-compiler` package. That will correspond to whichever version of `libprotobuf` you have installed on your system and should avoid the issues reported in the OP.

Downloading binaries from Google's site is nice, but seeing as there is a relatively tight dependency between compiler and library, not always a good idea.

DyumanAditya, 2021-03-12:
When I try to build abb_libegm and abb_libegm_samples, (using catkin_make_isolated on Windows) which are both in the same catkin_ws, I get the following error:
abb_libegm seems to build without error.

```
==> Processing catkin package: 'abb_libegm_samples'
==> Creating build directory: 'build_isolated\abb_libegm_samples'
==> Building with env: 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat'
==> cmake C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples -DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples -DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja in 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples'
-- The C compiler identification is MSVC 19.28.29337.0
-- The CXX compiler identification is MSVC 19.28.29337.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm_samples
-- Using CMAKE_PREFIX_PATH: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_state_publisher;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_service_provider;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_state_controller;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_hardware_interface;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_cpp_utilities;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_bringup_examples;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_sm_addin_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_msgs;c:\opt\ros\noetic\x64;C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_rws_managers;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_librws;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_libegm
-- This workspace overlays: C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_state_publisher;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_service_provider;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_state_controller;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_hardware_interface;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_cpp_utilities;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_bringup_examples;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_sm_addin_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_msgs;c:/opt/ros/noetic/x64
-- Found PythonInterp: C:/opt/ros/noetic/x64/python.exe (found suitable version "3.8.3", minimum required is "3")
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\ros\noetic\x64\lib\site-packages\em.py
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/noetic/x64/Scripts/nosetests.exe
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- Could NOT find abb_libegm (missing: abb_libegm_DIR)
-- Could not find the required component 'abb_libegm'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "abb_libegm" with
  any of the following names:

    abb_libegmConfig.cmake
    abb_libegm-config.cmake

  Add the installation prefix of "abb_libegm" to CMAKE_PREFIX_PATH or set
  "abb_libegm_DIR" to a directory containing one of the above files.  If
  "abb_libegm" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeError.log".
<== Failed to process package 'abb_libegm_samples':
  Command '['C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm\\env.bat', 'cmake', 'C:\\Users\\Dyuman\\catkin_ws1\\src\\abb_libegm_samples', '-DCATKIN_DEVEL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm_samples', '-DCMAKE_INSTALL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\install_isolated', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples' && 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat' cmake 'C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples' '-DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples' '-DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated' -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja

Command failed, exiting.
```