ros-industrial/abb_libegm #64: The question about EGM parameters

lijianhualeigua, 2019-11-22:
I viewer this answer  https://github.com/ros-industrial/abb_librws/issues/49 , and to test the EGM  parameters. I  find that whether  set PRG or PCG  to zero,  the will become the pure velocity control. Morever if you in pure velocity control, when you stop to send the targert velocity, the robot will continue the runing for a period of time , the value is equal the \CommTimeout of use EGMSetupUC instruction set. And chang the PPG notrequires to restart the IRC5 cointroller, but chang the PCG need to Pstart the IRC5 controller. 
In the process of testing the EGM parameters, I found some strange problems.
1: In the Technical reference manual RAPID instructions,Function and Datatypers.pdf about paramers  \PoseCorrGain  the value is between 0 and 1, but when i set the value more then 1, it didn't occur error. And have any way to test the separate effect of these two parameters on the EGM controller?
2: In the EGMActJoint instruction have a parameters \LpFilter, the desciption is low pass filter bandwith, used to filter sensor noise.   in the tesing ,i try to set this value 0 ,-100,-10000, it also din't occur error. The EGM working well, so i guess if the value is set \LpFilter <=0, it will be reset to default value, but i didnot know the default value about \LpFilter.
Thank you very much!

jontje, 2019-11-25:
> Morever if you in pure velocity control, when you stop to send the targert velocity, the robot will continue the runing for a period of time , the value is equal the \CommTimeout of use EGMSetupUC instruction set.

My recommended way of stopping a running EGM motion, is to send motion references so that the robot come to a standstill, and then use the RAPID `EGMStop` instruction to stop the EGM communication in an orderly way.

> And chang the PPG notrequires to restart the IRC5 cointroller, but chang the PCG need to Pstart the IRC5 controller.

Just to clarify, the `Proportional Position Gain` (found in the system configurations) requires a warmstart if changed, and the `Position Correction Gain` (of the RAPID `EGMRun` instructions) can be changed during runtime between two EGM communication sessions.

And, at least as far as I know, then the resulting position gain used during motions is `(Position Correction Gain)*(Proportional Position Gain)`.

> 1: In the Technical reference manual RAPID instructions,Function and Datatypers.pdf about paramers \PoseCorrGain the value is between 0 and 1, but when i set the value more then 1, it didn't occur error. And have any way to test the separate effect of these two parameters on the EGM controller?

I tested a RobotStudio simulation where I used `\PosCorrGain` larger than `1`, and it was accepted, but according to the manual this should not be the case. I will send a bug report to the EGM developers. Thanks for the note!

> 2: In the EGMActJoint instruction have a parameters \LpFilter, the desciption is low pass filter bandwith, used to filter sensor noise. in the tesing ,i try to set this value 0 ,-100,-10000, it also [...]

I will pass this along as well.

If I remember correctly, then the default value is `20 Hz`. I think the `\LpFilter` is more important when, for example, using `EGMSetupAI`, `EGMSetupAO` or `EGMSetupGI`. I.e. for using EGM with IO-signals connected to the robot controller's IO-board, but I haven't used this variant of EGM myself.

lijianhualeigua, 2019-11-26:
Great, thanks a lot.

> If I remember correctly, then the default value is 20 Hz.
In the <<Technical reference manual RAPID instructions,Function and Datatypers.pdf >> i didnot to find this value. but in the <<System parameters RW .pdf>> i find the parameters ext_motion_filter_bandwidth (found in the system configurations) the default value is 20.0 Hz, so maybe you remembe is this default value, not the  parameters \LpFilter default value in EGMActJoint instruction, if you find this default value description hope you can talk me. thanks.

> Just to clarify, the Proportional Position Gain (found in the system configurations) requires a warmstart if changed, and the Position Correction Gain (of the RAPID EGMRunJoint instructions) can be changed during runtime between two EGM communication sessions.

But when i change the Proportional Position Gain (found in the system configurations) in the real IRC5 controller, it didnot need to warmstart  and will with immediate effect. for example change this value to zeros, in the EGM motion will becomme pure velocity control, and that i change this value to 5, the next motion will become position control, it didnot to warmstart. and in the system configurations paper, when you change this value, system not to reminder you need to warmstart, but when you change the UDP ip address, the system will reminder you need to warmstart.

I find the low pass filter that used to filter sensor noise set in the EGMActJoint instruction  is not first order filter, it will maybe is second order filter or higher than second order . Because  i change the \LpFilter from 40  slowlly reduce to 0.065 HZ , and the position signal frquency is 0.125Hz. In the start the robot can following the desire trajectory , and then it will become overshooting and unstably if the \LpFilter is small then 1. If it is the first order low pass filter, use the control theory to analyise this system know that it not occur this phenomenon, so i guess this not first order filter. Do you have any file or other more detail information about this low pass filter hope you can talk me. thanks.
Thank you very much!

gavanderhoorn, 2019-11-26:
Even though this is an interesting discussion, I'm going to close the ticket: it's not reporting an issue with the package in this repository, and as such is more of a question.

That does not mean we cannot continue posting here @lijianhualeigua.

Please feel free to keep doing that.

(and the same for you of course @jontje ;))

closed 2019-11-26 by gavanderhoorn

jontje, 2019-11-26:
> In the <<Technical reference manual RAPID instructions,Function and Datatypers.pdf >> i didnot to find this value. but in the <<System parameters RW .pdf>> i find the parameters ext_motion_filter_bandwidth (found in the system configurations) the default value is 20.0 Hz, so maybe you remember is this default value, not the parameters \LpFilter default value in EGMActJoint instruction, if you find this default value description hope you can talk me. thanks.

As far as I know they are affecting the same parameter in the end. Just that `\LpFilter` overrides the default value from the system configurations. From the manual (for the `EGMActPose` instruction):

> [\LpFilter]
> Data type: num
> Low pass filter bandwidth, in Hertz (Hz), used to filter sensor noise.
> The default value is taken from the configuration of the EGMSetupXX instruction.

Since the `EGMSetupXX` instructions uses the previously mentioned system configuration, then it should be like that.

> But when i change the Proportional Position Gain (found in the system configurations) in the real IRC5 controller, it didn't need to warmstart and will with immediate effect.

You are correct, I just tried it out in RobostStudio and I was a bit surprised that it was like that. I have just been assuming a need for a restart, for that system configuration, because most of the configurations I use require a restart. 

> I find the low pass filter that used to filter sensor noise set in the EGMActJoint instruction is not first order filter, it will maybe is second order filter or higher than second order . Because i change the \LpFilter from 40 slowlly reduce to 0.065 HZ , and the position signal frequency is 0.125Hz. In the start the robot can following the desire trajectory , and then it will become overshooting and unstably if the \LpFilter is small then 1. If it is the first order low pass filter, use the control theory to analyse this system know that it not occur this phenomenon, so i guess this not first order filter. Do you have any file or other more detail information about this low pass filter hope you can talk me. thanks.

For my applications I haven't had a need to change the `\LpFilter` parameter, so I have little experience with the EGM filter. And, unfortunately, I don't have any additional documentation about the filter right now. Have you tried asking at the [ABB Robotics forum](https://forums.robotstudio.com/categories/rapid)?

lijianhualeigua, 2019-11-27:
Great, thanks a lot. I really appreciate it.

> Since the EGMSetupXX instructions uses the previously mentioned system configuration, then it should be like that.

You are correct, according my design experiment, i verification that \LpFilter overrides the default value from the system configurations. If the | \LpFilter didnot to set, then the system will use the value that set in system configurations. So in the  EGM controller only one low pass filter use to filter sensor noise.

lijianhualeigua, 2019-12-05:
About the EGM controller, i have serveral question. When use the EGM  to following desire trajectory, in order to improve the performace, we need to change the Proportional Position Gain and set it as possible as to high. But i foud that if the desire trajectory is a high frequency and need moving a large distance in joint space or cartesian space, when i increase the proportional position gain to 20 that is maximum value allowed, the track performance will be better but position deviation approximate to 2 mm, morever the robot always tremble during the movement. In the other hand exist 100 - 200 ms delay between send command position and receive position, the EGM system parameters ext_motion_level is use raw level and the ext_motion_filter_bandwidth is set to 100HZ.  If i use the speed feedforward and position feedback the delay will reduce to 30 - 50ms, the position deviation will reduce that also approximate to 1 mm. When i increase the run time for same trajectory the  performace will moe better and the position deviation can reduce to 0.1mm. 
      Have any way to improve the track performance when use the EGM to track trajectory that high frequency and need moving a large distance in joint space or cartesian space.  @jontje @gavanderhoorn .Thanks.

jontje, 2019-12-09:
Which `RobotWare` version are you using? Have you tried `6.10`? There have been some improvements there that might help you. From the release notes:

> EGM dynamic reduction of acceleration
> A new dynamic reduction of acceleration has been added to EGM that will help users avoiding event message 50375 "dynamic load too high" It will only reduce acceleration when needed.
> 
> A new field has added to egm.proto : utilizationRate that will give a value between 0-100 % if the client sends valid references.
> If the client sends too aggressive references EGM will reduce those and in the utilizationRate give values larger than 100%.
> The behavior of EGM will be best if this is avoided so users are encouraging to filter their speed and position references to avoid too high jerks and acceleration.

The `abb_libegm` library has not yet been updated to expose the new` utilizationRate` field, but you can still use the library with a `RobotWare 6.10` system.