ros-industrial/abb_libegm #66: Egm_interface is not connected, always "wait for an EGM communication session to start......"

StevenNUS, 2019-12-11:
I am using the abb_libegm_samples on a real robot of ABB IRB360. I tried several examples in the samples. But it seems the egm_interface is not connected. It always showed that "wait for an EGM communication session to start", even though I have started the RAPID codes on the controller. I have successfully configured the transmission protocol in RobotStudio, which has tested on rpi_abb_irc5. I can read the joint and position data via rpi_abb_irc5, based on the same configuration. But I do not know why it does not work for the abb_libegm_samples. Any help is much appreciated. 

The version of StateMachine Add-In is 1.1. The RobotWare is 6.09.00.01.
The EGM code is listed as:

MODULE TRobMain

    ! Home position.
    LOCAL CONST jointtarget home := [[0.49, -17.68, 4.91, -41.38, 0, 0], [9E9, 9E9, 9E9, 9E9, 9E9, 9E9]];
    
    ! Identifier for the EGM correction.
    LOCAL VAR egmident egm_id;
    
    ! Limits for convergance.
    LOCAL VAR egm_minmax egm_condition := [-0.1, 0.1];
    CONST num STATE_IDLE              := 0;
    CONST num STATE_INITIALIZE        := 1;
    CONST num STATE_RUN_RAPID_ROUTINE := 2;
    CONST num STATE_RUN_EGM_ROUTINE   := 3;
    VAR num current_state := STATE_INITIALIZE;
    
    ! Error numbers.
    VAR errnum CHANGE_STATE := -1;

    PROC main()
        EGMReset egm_id;
        EGMGetId egm_id;
        
        MoveAbsJ home, v200, fine, tool0;
        
        ! Setup the EGM communication.
        EGMSetupUC ROB_1, egm_id, "ROB_1", "ROB_1", \Joint \CommTimeout:=10;
        
        ! Prepare for an EGM communication session.
        EGMActJoint egm_id
                    \J1:=egm_condition
                    \J2:=egm_condition
                    \J3:=egm_condition
                    \J4:=egm_condition
                    \J5:=egm_condition
                    \J6:=egm_condition
                    \MaxPosDeviation:=100.0
                    \MaxSpeedDeviation:=100.0;
                    
        WHILE TRUE DO    
            ! Start the EGM communication session.
            EGMRunJoint egm_id, EGM_STOP_HOLD, \J1 \J2 \J3 \J4 \J5 \J6 \CondTime:=10 \RampOutTime:=10;
            WaitTime 2;
        ENDWHILE
        
    ERROR
        IF ERRNO = ERR_UDPUC_COMM THEN
            TPWrite "Communication timedout";
            TRYNEXT;
        ENDIF
    ENDPROC
ENDMODULE

I have tried several similar RAPID codes. But they always failed due to egm_interface is not connected.

jontje, 2020-01-17:
Hi @StevenNUS,

Sorry for the slow reply! Your question came at an unfortunate timing due to end-of-year processes and vacation time.

FlexPicker robots are unfortunately not supported by the EGM interface (only six-axis and seven-axis robots are currently supported). The robot controller's event log should show a message for that.

Also, how did you get an EGM license for the IRB360 robot? I am just curious to know.

StevenNUS, 2020-01-22:
Hi @jontje,

Thanks for your reply. We sent a quotation to ABB@Singapore at around June 2019. We got the FlexPicker at around 15th Oct 2019. We saw there was an EGM option in the specifications. So we chose the EGM option. We have attended two days training from ABB. But they have not mentioned EGM. I have checked with the ABB engineer. He has not heard the EGM. So I came here to download the library to try to set up the EGM on our new FlexPicker. But unfortunately the C++ always showed the message of "wait for an EGM communication session to start". I can install the EGM through StateInMachine. But it does not work.

gavanderhoorn, 2020-01-22:
@StevenNUS: could you perhaps copy the specific part about EGM from your specs here in a comment? We don't need a full copy, or a screenshot. It should just a single line.

I'd also be interested in how EGM got even mentioned in the context of FlexPickers.

jontje, 2020-01-22:
Just a couple of minor clarifications.

> So I came here to download the library to try to set up the EGM on our new FlexPicker. But unfortunately the C++ always showed the message of "wait for an EGM communication session to start".

The example code will wait until an EGM communication session has been setup, but that will never happen in your case (since EGM doesn't support your robot type). I.e. the the robot controller will not initiate the EGM communication channel.

> I can install the EGM through StateInMachine. But it does not work.

The StateMachine Add-In doesn't install EGM. During installation of a robot controller system, then the Add-In checks if the EGM license is present or not. If it is, then the Add-In loads a RAPID module with RAPID code intended for easing the use of EGM. It's the license file that indicates if EGM is "installed" or not.

I am guessing that there has probably been some unfortunate misunderstanding during the quotation request. Since, as far as I know, then EGM has never had support for FlexPicker robots, but I can of course be wrong.

gavanderhoorn, 2020-01-31:
I'm going to close this.

EGM is not officially supported on FlexPickers, so the timeout reported by @StevenNUS is expected.

@StevenNUS: I'd still be interested in knowing some more about your system, so if you ever find out, please let us know.

For now this is closed.

Feel free to keep commenting on the issue of course though.

closed 2020-01-31 by gavanderhoorn

StevenNUS, 2020-01-31:
@gavanderhoorn Sorry for the late reply due to Lunar New Year Holiday. I think the code is the same one with others. When you install the EGM through StateInMachine. It will automatically upload some new codes, perhaps 4 or 5 modules. I think it is just the sample codes and it is the same with all the robots. I will upload the codes soon.

gavanderhoorn, 2020-01-31:
I was referring to how you got EGM licensed on your FlexPicker :)

StevenNUS, 2020-01-31:
@jontje 

- The example code will wait until an EGM communication session has been setup, but that will never happen in your case (since EGM doesn't support your robot type). I.e. the the robot controller will not initiate the EGM communication channel.

I saw the IRC5 controller is supported by the EGM. So I tried the setup because the FlexPicker is also used the IRC5 controller. But it does not work.

- The StateMachine Add-In doesn't install EGM. During installation of a robot controller system, then the Add-In checks if the EGM license is present or not. If it is, then the Add-In loads a RAPID module with RAPID code intended for easing the use of EGM. It's the license file that indicates if EGM is "installed" or not.

I am guessing that there has probably been some unfortunate misunderstanding during the quotation request. Since, as far as I know, then EGM has never had support for FlexPicker robots, but I can of course be wrong.

After the installation of EGM from StateInMachine, the system has automatically uploaded some codes, perhaps 4 or 5 modules or even more. I am not sure. But it does not work. I saw the EGM options from RoboStudio (seems under the "license", I forgot the name.). I think the uploaded codes is the same one with all the robots.

StevenNUS, 2020-01-31:
@gavanderhoorn 
I saw the EGM options from RoboStudio (seems under the "license" or what, I forgot the name.). So I install from StateInMachine and I got the sample codes. But it does not work. I will try to take a picture from the RoboStudio tomorrow. I saw the IRC5 controller is supported by EGM so I tried this operation (the FlexPicker is also used this controller). But it seems the FlexPicker is not supported by EGM.

StevenNUS, 2020-02-06:
Hi @gavanderhoorn, 

I have uploaded three pictures from the screenshot to show the configuration of our FlexPicker. These pictures are captured from the  RobotStudio. Thanks.

[image]
[image]
[image]

gavanderhoorn, 2020-02-06:
@jontje: any idea how this happened? Seems like someone made a mistake somewhere when provisioning the controller/licenses?

jontje, 2020-02-07:
> @jontje: any idea how this happened? Seems like someone made a mistake somewhere when provisioning the controller/licenses?

No idea. And, yes, I think it is probable that a mistake happened somewhere.

CaptainNUS, 2020-02-27:
Hi @gavanderhoorn @jontje:

May I check if there is some other libraries or techniques to support sending trajectories to the FlexPicker robot? Right now I can only program with RAPID codes for robot control.

I used Python codes to get the desired position information and sent to the FlexPicker through socket. But I don't know how to do the motion planning with Python or C++ in Ubuntu system.

Any help is much appreciated. Thanks.

jontje, 2020-03-02:
@CaptainNUS, I created a new issue for your latest question since I see it as a separate issue from this one, and it may also help others looking for the same thing.