ros-industrial/abb_libegm #72: "No syntax specific" warning when compiling protobuf messages

gavanderhoorn, 2020-02-03:
We may want to add a `syntax` key to the `.proto` files we ship here ([these](https://github.com/ros-industrial/abb_libegm/tree/59b6b94cf1cc79135f05fc2b43dbacb625172e78/proto)):

```
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: egm.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
```

gavanderhoorn, 2020-02-03:
@jontje: we can do this for the `.proto`s we added for `abb_libegm` in this repository, but the main `egm.proto` is not maintained here.

That doesn't mean we cannot add the key, but it would be nice if "upstream" could do it as well.

jontje, 2020-02-04:
> That doesn't mean we cannot add the key, but it would be nice if "upstream" could do it as well.

I have mentioned it to the EGM developers.

I will add the syntax specifier to `egm.proto` when I update for the newest protocol version.

PR #78 (2020-02-07): Update proto files (fix #72)
This PR updates the EGM proto file to the latest version, which adds a field for the robot's utilization rate.

A syntax specifier has also been added.

This fixes #72.

Finally, the EGM wrapper proto file has been updated to include the new field.
commit 5d0d31a2 (2020-02-07): Merge pull request #78 from jontje/update_proto_files

Update proto files (fix #72)
closed 2020-02-07 by jontje