ros-industrial/abb_libegm #73: Undefined Load, Event Message 41438

jbeck28, 2020-02-04:
I've had EGM working on my desktop for about 6 months now, and I've decided to offload running the robot from my desktop to another computer. I copied the program I've been running (which essentially is a modification of [https://github.com/kth-ros-pkg/yumi](url)) and I find that when I run the program, the state machine add-in gets to line 326 in the program TRobEGM, when it setsup EGMActJoint, I get that an error on the controller saying that the load is undefined. I've never had this issue before and the only thing that changed is I downloaded the latest version of this package, and modified my CMake file to reflect that this is no longer a catkin package. 

Is there any change to this package which may have caused this? I see that `egm_load` has a 0kg load, defined on line 151 of the same module. I'm not exactly sure if it's safe to modify this, and it also seems likely that it was set to `load0` on line 305 (which is .001kg). 

I haven't followed the flow of the StateMachineAddIn enough to see if this is the case, or if I should just modify the `egm_load` defined in the addin... but since I didn't have this issue before, I thought I'd post it here. 

Any assistance would be highly appreciated!

gavanderhoorn, 2020-02-04:
@jbeck28: could you please always provide links to lines you are referring to? And please use perma-links (ie: press <kbd>y</kbd>) instead of regular ones, as line numbers can and do change over time.

jontje, 2020-02-04:
@jbeck28, are you also using the latest [abb_librws](https://github.com/ros-industrial/abb_librws)? I know of a bug that has crept into the master `branch` (which affects nestled RAPID records), and this could be the cause. I have a [fix](https://github.com/jontje/abb_librws/commit/196582c70d2d9778b4fb7ea383ac2dbb510f2442) that you can try out, if this is the case.

Otherwise, there should be a RAPID procedure called `parseEGMSettings` in the `TRobEGM` module, and you can manually set the values of this section:
```
! Arguments for EGMAct instructions
egm_tool := settings.activate.tool;
egm_load := egm_tool.tload;
egm_wobj := settings.activate.wobj;
```

For example, just to test, to: 
```
! Arguments for EGMAct instructions
egm_tool := tool0;
egm_load := egm_tool.tload;
egm_wobj := wobj0;
```

jbeck28, 2020-02-04:
@gavanderhoorn 
I should have been more clear, my apologies! All of those line numbers are in reference to specific modules which make up the StateMachine Add-In. I'm not sure where to find source code other than once it is installed on the controller. 

@jontje 

I actually did not use the latest version of abb_librws, perhaps I should try to install that. 

In the mean time I'll try those modifications that you mentioned. 

Thank you for your reply

jontje, 2020-02-04:
> @gavanderhoorn
> I should have been more clear, my apologies! All of those line numbers are in reference to specific modules which make up the StateMachine Add-In. I'm not sure where to find source code other than once it is installed on the controller.

The RAPID modules are currently only available in the distribution format, i.e. packaged into [v1.0](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee) and [v1.1](https://robotapps.robotstudio.com/#/viewApp/c163de01-792e-4892-a290-37dbe050b6e1). But, yes, it would be nice to make the RAPID modules available directly as well.

jbeck28, 2020-02-06:
The simple modification you mentioned worked perfectly. Thank you @jontje!

jontje, 2020-02-07:
> The simple modification you mentioned worked perfectly. Thank you @jontje!

@jbeck28, that's great! Which of the suggestions did you use? The modification of the RAPID code, or the fix for [abb_librws](https://github.com/ros-industrial/abb_librws)?

jbeck28, 2020-02-17:
I never replied, my apologies! I the modification of the rapid code fixed my issue:
```

! Arguments for EGMAct instructions
egm_tool := tool0;
egm_load := egm_tool.tload;
egm_wobj := wobj0;
```

jontje, 2020-02-18:
> I never replied, my apologies! The modification of the rapid code fixed my issue:

No worries @jbeck28, and thanks for letting me know.

I will close this issue, but feel free to post additional comments if necessary.

closed 2020-02-18 by jontje