ros-industrial/abb_libegm #77: Host State Machine addin somewhere

gavanderhoorn, 2020-02-07:
Tracking this here, even though it's not really the place.

@jontje wrote (in https://github.com/ros-industrial/abb_libegm/issues/73#issuecomment-581956109):
> The RAPID modules are currently only available in the distribution format, i.e. packaged into [v1.0](https://robotapps.robotstudio.com/#/viewApp/7fa7065f-457f-47ce-98d7-c04882e703ee) and [v1.1](https://robotapps.robotstudio.com/#/viewApp/c163de01-792e-4892-a290-37dbe050b6e1). But, yes, it would be nice to make the RAPID modules available directly as well.

traversaro, 2020-02-08:
fyi @AlbertoRemusIIT

jontje, 2020-02-10:
@gavanderhoorn, I updated the title just for clarification, since the StateMachine Add-In is already hosted at the [ABB RobotApps](https://robotapps.robotstudio.com/#/landing) site (in distribution format).

gavanderhoorn, 2020-02-10:
Just to be clear: we'd be hosting the addin itself and the accompanying RAPID code (templates), correct?

jontje, 2020-02-17:
> Just to be clear: we'd be hosting the addin itself and the accompanying RAPID code (templates), correct?

All the Add-In’s insides could be hosted, but I don’t see much sense in including the packaged format (that is generated from the “Add-In Packaging Tool 1.9” found [here]( https://new.abb.com/products/robotics/robotstudio/downloads)). And with insides I mean the RAPID code, system configurations and installation script.

gavanderhoorn, 2020-03-23:
@jontje: any update here?

jontje, 2020-03-24:
> @jontje: any update here?

Not as of yet, and I don't have an estimate of when I could work on this.