ros-industrial/abb_libegm #82: Add support for RobotWare versions 6.0 - 6.06.03

jontje, 2020-02-19:
When PR #81 has been merged it would be possible to re-add support for the currently incompatible RobotWare versions 6.0 - 6.06.03.

gavanderhoorn, 2020-02-19:
Hm. That didn't work as I had expected.

But great work with #81.

gavanderhoorn, 2020-11-02:
@jontje: I've 'inverted' the title of this issue to better reflect the current state of things.

That turns this issue into an actual issue, instead of an enhancement request.

I'll also pin the issue, so users looking at the issue tracker would immediately see it.

gavanderhoorn, 2023-02-22:
As `6.06` is really rather old now, I'm going to close this -- but keep it pinned.

If/when someone really wants to interface with such an old RW version, this issue could be used to track any related development.

closed 2023-02-22 by gavanderhoorn