ros-industrial/abb_libegm #85: Sending trajectories to robots without EGM support

jontje, 2020-03-02:
Seperate issue for @CaptainNUS's question from https://github.com/ros-industrial/abb_libegm/issues/66#issuecomment-591841601:

> Hi @gavanderhoorn @jontje:
> 
> May I check if there is some other libraries or techniques to support sending trajectories to the FlexPicker robot? Right now I can only program with RAPID codes for robot control.
> 
> I used Python codes to get the desired position information and sent to the FlexPicker through socket. But I don't know how to do the motion planning with Python or C++ in Ubuntu system.
> 
> Any help is much appreciated. Thanks.

jontje, 2020-03-02:
> May I check if there is some other libraries or techniques to support sending trajectories to the FlexPicker robot? Right now I can only program with RAPID codes for robot control.

I would use [RWS](http://developercenter.robotstudio.com/webservice/api_reference) via `abb_libegm`'s companion library [abb_librws](https://github.com/ros-industrial/abb_librws).

For example by:

1. Creating/generating the path/trajectory you want to follow.
2. Generate a RAPID module, with a RAPID procedure for running the path/trajectory.
3. Upload the module to the robot controller.
4. Load the module into the RAPID program.
5. Execute the newly loaded procedure.

If you are using `abb_librws` in combination with the RobotWare [StateMachine Add-In](https://robotapps.robotstudio.com/#/viewApp/c163de01-792e-4892-a290-37dbe050b6e1), then you can for example do something like this:

```
abb::rws::RWSStateMachineInterface rws_interface("127.0.0.1");

// Home position (for an IRB360_1_1130_4D_STD_03).
std::string rt_home = "[[0,0,-937.463],[0,1,0,0],[0,0,0,0],[9E+9,9E+9,9E+9,9E+9,9E+9,9E+9]]";

//--------------------------------------------
// Create mockup points for a path
//--------------------------------------------
abb::rws::RobTarget rt_1;
abb::rws::RobTarget rt_2;
abb::rws::RobTarget rt_3;
rt_1.parseString(rt_home);
rt_2.parseString(rt_home);
rt_3.parseString(rt_home);

rt_2.pos.x.value += 200.0;
rt_2.pos.y.value += 200.0;

rt_3.pos.x.value += 200.0;
rt_3.pos.y.value += 200.0;
rt_3.pos.z.value -= 200.0;

//--------------------------------------------
// Create a RAPID module
//--------------------------------------------
std::stringstream ss;
ss << "MODULE MockupPath(SYSMODULE)\n"
   << "\tPROC Path0()\n"
   << "\t\tMoveL " << rt_1.constructString() << ", v300, z30, tool0;\n"
   << "\t\tMoveL " << rt_2.constructString() << ", v300, z30, tool0;\n"
   << "\t\tMoveL " << rt_3.constructString() << ", v300, z30, tool0;\n"
   << "\t\tMoveL " << rt_2.constructString() << ", v300, z30, tool0;\n"
   << "\t\tMoveL " << rt_1.constructString() << ", v300, fine, tool0;\n"
   << "\tENDPROC\n"
   << "ENDMODULE";

//--------------------------------------------
// Upload, load, run and unload the module
//--------------------------------------------
rws_interface.uploadFile(abb::rws::RWSClient::FileResource("MockupPath.sys"), ss.str());
std::this_thread::sleep_for(std::chrono::milliseconds(500));

rws_interface.services().rapid().runModuleLoad("T_ROB1", "HOME:/MockupPath.sys");
std::this_thread::sleep_for(std::chrono::milliseconds(500));

rws_interface.services().rapid().runCallByVar("T_ROB1", "Path", 0);
std::this_thread::sleep_for(std::chrono::milliseconds(5000));

rws_interface.services().rapid().runModuleUnload("T_ROB1", "HOME:/MockupPath.sys");
```

Here is a video showing a RobotStudio simulation when running the code: [IRB360 trajectory via abb_librws.zip](https://github.com/ros-industrial/abb_libegm/files/4277232/IRB360.trajectory.via.abb_librws.zip). Note how the left side shows how the RAPID module is loaded during runtime.

And you can of course use `MoveJ` instead of `MoveL`, different tools, zones and specifying the time  to reach each point etc.

gavanderhoorn, 2020-03-23:
As this is not really an issue with the package in this repository, I'm going to close it.

Feel free to keep commenting on this issue of course.

I hope @CaptainNUS was able to benefit from the comment by @jontje.

closed 2020-03-23 by gavanderhoorn