ros-industrial/abb_libegm #90: How to properly cite the abb_libegm project

traversaro, 2020-04-22:
Hi @jontje, 
   I was wondering if there is any specific publication/work that would like user to cite when they write a academic pubblication, and they want to properly refer/cite `abb_libegm` ? 

Thanks in advance!

jontje, 2020-04-23:
Hi @traversaro,

Good question.

A start would be to at least cite that ABB developed `abb_libegm` with support from the EU projects
[SYMBIO-TIC](http://www.symbio-tic.eu/) and [ROSIN](https://rosin-project.eu/) (`abb_libegm` wouldn't exist without them). See the [readme's acknowledgements section](https://github.com/ros-industrial/abb_libegm#acknowledgements) for some more details.

I hope this helps!

traversaro, 2020-04-23:
Hi  @jontje , thanks! 

Typically in academic papers we  don't cite the funding of the work/software that we refer to (as otherwise it may quite complex to track and mention the funding of all the papers that you cite), and this job is typically left to the resources that are  linked by the citation. 

In any case, for the citation I was thinking something like: 
~~~
@inproceedings{tjerngren2018ease,
  title={Ease-of-Use Packages between ROS and ABB Robots},
  author={Tjerngren, Jon and ABB},
  booktitle={ROS-Industrial Conference},
  url={https://rosindustrial.org/s/20181212_Jon_Tjerngren.PDF},
  year={2018}
}
~~~
(That is already cited by two pubblication, see https://scholar.google.com/scholar?hl=en&as_sdt=0%2C5&q=abb_libegm&btnG= ) . 

Otherwise one can cite the raw GitHub repo, something like: 
~~~
@misc{abblibegm2018,
  author = {Tjerngren, Jon and ABB},
  title = {abb_libegm},
  year = {2020},
  publisher = {GitHub},
  journal = {GitHub repository},
  howpublished = {\url{https://github.com/ros-industrial/abb_libegm}},
  commit = {c91601dfc9844f2d37b1eac0d2b7060f0ac2550b}
}
~~~
but I did not checked how this get rendered in the templates that I used, I just copied it from https://academia.stackexchange.com/questions/14010/how-do-you-cite-a-github-repository .

jontje, 2020-04-23:
> Typically in academic papers we don't cite the funding of the work/software that we refer to (as otherwise it may quite complex to track and mention the funding of all the papers that you cite), and this job is typically left to the resources that are linked by the citation.

Ah, yes, I have some vague memories of having heard that before :sweat_smile:

gavanderhoorn, 2020-04-23:
I'm not aware of any publications. There is no seminal paper which describes `abb_libegm`.

My suggestion would be to use the first option you mention (ie: @jontje's presentation at the ROS-I Conference 2018).

traversaro, 2020-04-23:
I think I can close  the  issue as I gather the information I was interested  in, thanks!

closed 2020-04-23 by traversaro