ros-industrial/abb_libegm #104: Stopping cartesian velocity EGM session

shobhitcito, 2020-07-20:
I am streaming velocity set points to the ABB IRB 1200 robot (IRC5 Robotware 6.10) via EGM. How can I make sure that the robot stop executing the velocity set point given after certain time if I lose connection to the server (PC generating set points)?

Currently, the robot continues to execute the last velocity set point it received in case if the client loses connection to the server. This is extremely important to handle this possible situation in operation.

We currently use:
`
EGMRunPose EgmStopHold\noWaitCond \X\Y\Z \CondTime:=0.5, \RampInTime=0.5\PosCorrGain=0;
`

The response I would likt to have:

** Just as in position setpoint streaming, condtime defines the convergence time after which it will stop the robot from movement, I would like to define a time after which the robot stop executing velocity setpoint given**

jontje, 2020-08-12:
Hello @shobhitcito,

> I am streaming velocity set points to the ABB IRB 1200 robot (IRC5 Robotware 6.10) via EGM. How can I make sure that the robot stop executing the velocity set point given after certain time if I lose connection to the server (PC generating set points)?

Have you specified the optional `\CommTimeout` argument when setting up the EGM communication channel? E.g. with the `EGMSetupUC` instruction.

If I remember correctly, then the default timeout is 1 second.

gavanderhoorn, 2020-08-25:
I'm going to close this due to inactivity.

@shobhitcito: if this is still something you'd wish to discuss or continue, please comment and we can re-open.

closed 2020-08-25 by gavanderhoorn