ros-industrial/abb_libegm #107: How to use EGMPlanner to controller external axis

lijianhualeigua, 2020-09-23:
Through this issue [https://github.com/ros-industrial/abb_libegm/issues/57] , I found that the external axis is controlled through EGM, but I have a question. For using EGM to control the external axis, it is possible to setting different mechanical units through EGMSetupUC, and make corresponding connections through different UDP ports. But I see that EGMPlanner message contains External Joint information, and i can use the set_allocated_externaljoints() to set external joint value, but EGMActJoint  only can set six joint
 `EGMActJoint EGMid [\Tool] [\WObj] [\TLoad] [\J1] [\J2] [\J3] [\J4]
[\J5] [\J6] [\LpFilter] [\SampleRate] [\MaxPosDeviation]
[\MaxSpeedDeviation].` 
So I have a question, how to use the external joint infomation, and how to use EGMPlanner message to control external axis?  Because the robot and the external axis belong to different mechanical units.

Thanks in advance,

jontje, 2020-09-23:
Hi @lijianhualeigua, can you provide a brief description of your system? For example, `RobotWare` version and options, as well as the mechanical units and RAPID tasks that are configured in the system.

> So I have a question, how to use the external joint infomation, and how to use EGMPlanner message to control external axis? Because the robot and the external axis belong to different mechanical units.

You need to have different RAPID tasks for each mechanical unit you want to control with EGM (see https://github.com/ros-industrial/abb_libegm/issues/57#issuecomment-526475159). To have different RAPID tasks, then you need the `RobotWare` option `623-1 Multitasking`.

When you have different RAPID tasks, then you can use `EGMSetupUC` to set up EGM channels for respective mechanical unit, and `EGMActJoint` is used to configure each channel.

lijianhualeigua, 2020-09-23:
@jontje Thank you !  I upload four pictrue about the system and task configure infomation.
The system infomation:

[image]

The task infomation:
[image]

The mechanical unit 
[image]

The mechanical unit group
[image]

If i need use one motion task and EGM UDP port  to controller the robot and external axis, how can do this, how to use EGMPlanner message.

Thanks in advance.

jontje, 2020-09-23:
# Robot Controller

Your system configurations look OK, except maybe that the mechanical unit group `Exter_axis` should probably have the `M7DM1` unit under the `Mechanical Unit 1` column instead of the `Robot` column. The `Robot` column is for robots with a TCP (according to the manual).

Then you need to set up the RAPID code with something like this:

- The `T_ROB1` task:
   ```
   PROC main()
       VAR egmident egm_id;   
       EGMGetId egm_id;

       ! "UCdevice_ROB_1" needs to be configured in the system configurations.
       EGMSetupUC ROB_1, egm_id, "default", "UCdevice_ROB_1" \Joint;
       EGMActJoint egm_id;
        
       EGMRunJoint egm_id, EGM_STOP_HOLD \J1 \J2 \J3 \J4 \J5 \J6;
        
       EGMReset egm_id;
   ENDPROC
   ```

- The `M7` task:
   ```
   PROC main()        
       ! Make sure that the external axis unit is active.
       ActUnit M7DM1;

       VAR egmident egm_id;
       EGMGetId egm_id;

       ! "UCdevice_M7" needs to be configured in the system configurations.
       EGMSetupUC M7DM1, egm_id, "default", "UCdevice_M7" \Joint;
       EGMActJoint egm_id;

       EGMRunJoint egm_id, EGM_STOP_HOLD;
        
       EGMReset egm_id;
   ENDPROC
   ```

# External Computer

For the EGM servers, then you can do something like this:

```
boost::asio::io_service io_service;
boost::thread_group worker_threads;

abb::egm::BaseConfiguration configuration;

// Assuming that the robot has been configured to use port 6510. 
configuration.axes = abb::egm::RobotAxes::Six;
abb::egm::EGMControllerInterface egm_interface_1(io_service, 6510, configuration);

// Assuming that the external axis has been configured to use port 6511.
configuration.axes = abb::egm::RobotAxes::None;
abb::egm::EGMControllerInterface egm_interface_2(io_service, 6511, configuration);

while(true)
{
  if(egm_interface_1.waitForMessage(500))
  {
    // egm_interface_1.read(...);
    // ...
    // egm_interface_1.write(...);
  }

  if(egm_interface_2.waitForMessage(500))
  {
    // egm_interface_2.read(...);
    // ...
    // egm_interface_2.write(...);
  }
}
```

lijianhualeigua, 2020-09-23:
@jontje  Thank you for your patient answers.
But you use two EGM server to controller robot and external axis, this way not need to allocat external joint value in EGMPlanner message. So  this way can not realize the sync  move about robot and external axis by reason of  not use only one EGM server . If use MoveJ\MoveL to move robot and external axis, it can realize the sync move about robot and external axis.  I curious the function about external joint in the EGMPlanner message, it  just only to control the seven axis robot? 

Thanks in advance.

jontje, 2020-09-23:
Unfortunately, as far as I am aware, then it is not possible to control both the robot and the external axis via the same EGM channel. 

It's a limitation in the EGM interface, and I haven't seen any updates about it in the latest `RobotWare` release notes.

lijianhualeigua, 2020-09-23:
@jontje Thank you for your patient answers.
I have other question, if we have three external axis and this belong three mechanical unit, according you answer i need to use four EGM server  channel to controller robot and  other three external axis. 
If  we have three external axis and this belong one mechanical unit, so i need use two EGM server  channel  to controller robot and  other three  external axis.  SO i have conclusion that one EGM server channel only corresponding one  mechanical unit, i don't konw this conclusion whether  is right.
Whether i can to compond the three mechanical unit to one mechanical unit, just like mechanical unit group , and use EGMSetupUC to set this mechanical unit group to contol the robot and external axis move.

Thanks in advance.

jontje, 2020-09-24:
> @jontje Thank you for your patient answers.

No worries, I am happy to help.

> I have other question, if we have three external axis and this belong three mechanical unit, according you answer i need to use four EGM server channel to controller robot and other three external axis.

Yes.

Also, another limitation of EGM is that it can only manage 4 active channels at the same time (if I remember correctly).

> If we have three external axis and this belong one mechanical unit, so i need use two EGM server channel to controller robot and other three external axis.

Yes.

> SO i have conclusion that one EGM server channel only corresponding one mechanical unit, i don't konw this conclusion whether is right.

That is correct.

> Whether i can to compond the three mechanical unit to one mechanical unit, just like mechanical unit group , and use EGMSetupUC to set this mechanical unit group to contol the robot and external axis move.

I have never attempted anything like that, so I don't know the feasibility of this approach. But if you can get all the external axes controlled by the same mechanical unit, then it should work with one EGM channel for the compounded mechanical unit.

I suggest that you post a question about it on the official [ABB Robotics User Forum](https://forums.robotstudio.com/), or that you contact your local `ABB Robotics` sale organization.

gavanderhoorn, 2020-11-02:
I'm going to assume this was answered, and as this does not seem like something which needs changes in `abb_libegm` itself, I'm going to close it.

Feel free to keep commenting on the issue of course @lijianhualeigua.

And it would also be interesting to hear about your experiences with setting up the multi-group system with EGM.

closed 2020-11-02 by gavanderhoorn

Didier-Surface, 2021-03-05:
@lijianhualeigua could you contact me in private message? I would like to discuss about this issue