ros-industrial/abb_libegm #118: support for the Omnicore controller

marekcygan, 2021-02-11:
I would like to ask whether this library will work with the new controller Omnicore (e.g., for IRB 1300), which also requires Robotware 7+ if I understand correctly.

gavanderhoorn, 2021-02-11:
This library: afaik, yes, it is compatible with RW7. @jontje has tested this. He'll probably confirm tomorrow.

`abb_librws` however is not: RWS changed a bit between RW6 and RW7. See also https://github.com/ros-industrial/abb_librws/issues/80.

jontje, 2021-02-12:
I have tested abb_libegm with single armed YuMi robots, which runs on OmniCore controllers, and it worked fine.

I have not tested with the latest RobotWare `7.1.0` version, but I don't expect any problems (based on what I can see in the release notes).

Also, I recommend to setting a test with RobotStudio for the case in question, since if that works then it will most likely work with a real robot.

gavanderhoorn, 2021-02-12:
Seeing as the immediate question was answered, I'm going to close this.

@marekcygan: if it turns out something does need to be fixed or changed in `abb_libegm`, we can track it in more on-topic issues.

closed 2021-02-12 by gavanderhoorn

marekcygan, 2021-02-12:
@jontje @gavanderhoorn thanks for quick answers!