ros-industrial/abb_libegm #120: abb_libegm vs abb_driver

stevedanomodolor, 2021-03-06:
this is out of curiosity, 
What are the advatatges of using the abb_libegm vs using the abb_driver?

traversaro, 2021-03-06:
`abb_libegm`  is a pure C++ library that do not depend on ROS, while the `abb_robot_driver` provides ready to use ROS nodes.

gavanderhoorn, 2021-03-06:
> abb_libegm vs abb_driver

`abb_driver` is not the same as `abb_robot_driver`.

Could you clarify which one you're asking about @stevedanomodolor?

stevedanomodolor, 2021-03-06:
abb_robot_driver, @gavanderhoorn, 
If i wanted to use moveit for motion planning, would you advice me to use dirrect abb_robot_controller or send the command directly with the abb_lib_egm?

gavanderhoorn, 2021-03-07:
> What are the advatatges of using the abb_libegm vs using the abb_driver?

I believe @traversaro explained it well enough: `abb_robot_driver` is an actual ROS *driver* for ABB controllers. `abb_libegm` is a low-level library for interfacing with EGM capable ABB controllers.

The former uses the latter, but they are definitely not comparable.

> If i wanted to use moveit for motion planning, would you advice me to use abb_robot_driver or send the command directly with the abb_lib_egm?

Depends on how much work you want to do yourself.

Do you want to simply execute trajectories? Use `abb_robot_driver`.

Do you first want to have to write your own action server, trajectory interpolator and execution node(s)? Then use `abb_libegm` (probably in combination with `abb_librws`.

> Or it there already a way of doing motion plannig directlty with the abb_robot_driver?

I'm not sure I understand what you mean by this.

Drivers typically don't perform any "motion planning". They are components which essentially transform data and control flow:

 - from OEM-specific representations and conventions to ROS, and
 - from ROS-specific representations and conventions to OEM

depending on what you'd classify as *motion planning*, I don't believe that would fall under these two responsibilities.

gavanderhoorn, 2021-03-07:
As this is not an issue with `abb_libegm`, but more of a usage / general question, I'm going to close it.

Feel free to keep commenting on it of course @stevedanomodolor.

closed 2021-03-07 by gavanderhoorn

stevedanomodolor, 2021-03-07:
@gavanderhoorn thank you very much , I understood it clearly