ros-industrial/abb_libegm #121: Can't build abb_libegm_samples on Windows

DyumanAditya, 2021-03-15:
Hi,
I found these abb_libegm samples from [issue #18](https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-457592204) and want to experiment with them to see if they can help me out. I have a `catkin_ws`  with only `abb_libegm` and `abb_libegm_samples`. When I run `catkin_make_isolated` `abb_libegm` builds successfully but `abb_libegm_samples` gives me this error:

<details><summary>Click to expand</summary>

```
==> Processing catkin package: 'abb_libegm_samples'
==> Creating build directory: 'build_isolated\abb_libegm_samples'
==> Building with env: 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat'
==> cmake C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples -DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples -DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja in 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples'
-- The C compiler identification is MSVC 19.28.29337.0
-- The CXX compiler identification is MSVC 19.28.29337.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm_samples
-- Using CMAKE_PREFIX_PATH: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_state_publisher;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_service_provider;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_state_controller;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_hardware_interface;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_cpp_utilities;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_bringup_examples;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_sm_addin_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_msgs;c:\opt\ros\noetic\x64;C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_rws_managers;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_librws;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_libegm
-- This workspace overlays: C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_state_publisher;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rws_service_provider;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_state_controller;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_hardware_interface;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_cpp_utilities;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_robot_bringup_examples;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_sm_addin_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_rapid_msgs;C:/Users/Dyuman/catkin_ws/devel_isolated/abb_egm_msgs;c:/opt/ros/noetic/x64
-- Found PythonInterp: C:/opt/ros/noetic/x64/python.exe (found suitable version "3.8.3", minimum required is "3")
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\ros\noetic\x64\lib\site-packages\em.py
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/noetic/x64/Scripts/nosetests.exe
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- Could NOT find abb_libegm (missing: abb_libegm_DIR)
-- Could not find the required component 'abb_libegm'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "abb_libegm" with
  any of the following names:

    abb_libegmConfig.cmake
    abb_libegm-config.cmake

  Add the installation prefix of "abb_libegm" to CMAKE_PREFIX_PATH or set
  "abb_libegm_DIR" to a directory containing one of the above files.  If
  "abb_libegm" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeError.log".
<== Failed to process package 'abb_libegm_samples':
  Command '['C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm\\env.bat', 'cmake', 'C:\\Users\\Dyuman\\catkin_ws1\\src\\abb_libegm_samples', '-DCATKIN_DEVEL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm_samples', '-DCMAKE_INSTALL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\install_isolated', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples' && 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat' cmake 'C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples' '-DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples' '-DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated' -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja

Command failed, exiting.
```
</details>

What could be going wrong?

DyumanAditya, 2021-03-15:
**UPDATE:** I read [here](https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-692121293) the `abb_libegm_samples` were made some time ago when `abb_libegm` was a catkin package.

I have updated my CMakeList.txt accordingly (I think, please correct me if I've done something wrong), here it is:

<details><summary>Click to expand</summary>

```cmake
cmake_minimum_required(VERSION 2.8.3)
project(abb_libegm_samples)

find_package(catkin REQUIRED COMPONENTS roscpp)
find_package(abb_libegm REQUIRED)

###################################
## catkin specific configuration ##
###################################
catkin_package(
  CATKIN_DEPENDS roscpp
  DEPENDS abb_libegm
)

###########
## Build ##
###########

include_directories(
 ${catkin_INCLUDE_DIRS}
)

#################################
# Build the basic level samples #
#################################

add_executable(${PROJECT_NAME}_joint_trajectory_node src/a1_joint_trajectory_node.cpp)
set_target_properties(${PROJECT_NAME}_joint_trajectory_node PROPERTIES OUTPUT_NAME a1_joint_trajectory_node PREFIX "")
add_dependencies(${PROJECT_NAME}_joint_trajectory_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_joint_trajectory_node ${catkin_LIBRARIES})

add_executable(${PROJECT_NAME}_pose_trajectory_node src/a2_pose_trajectory_node.cpp)
set_target_properties(${PROJECT_NAME}_pose_trajectory_node PROPERTIES OUTPUT_NAME a2_pose_trajectory_node PREFIX "")
add_dependencies(${PROJECT_NAME}_pose_trajectory_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_pose_trajectory_node ${catkin_LIBRARIES})

########################################
# Build the intermediate level samples #
########################################

add_executable(${PROJECT_NAME}_joint_static_goal_node src/b1_joint_static_goal_node.cpp)
set_target_properties(${PROJECT_NAME}_joint_static_goal_node PROPERTIES OUTPUT_NAME b1_joint_static_goal_node PREFIX "")
add_dependencies(${PROJECT_NAME}_joint_static_goal_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_joint_static_goal_node ${catkin_LIBRARIES})

add_executable(${PROJECT_NAME}_pose_static_goal_node src/b2_pose_static_goal_node.cpp)
set_target_properties(${PROJECT_NAME}_pose_static_goal_node PROPERTIES OUTPUT_NAME b2_pose_static_goal_node PREFIX "")
add_dependencies(${PROJECT_NAME}_pose_static_goal_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_pose_static_goal_node ${catkin_LIBRARIES})

####################################
# Build the advanced level samples #
####################################

add_executable(${PROJECT_NAME}_joint_controller_node src/c1_joint_controller_node.cpp)
set_target_properties(${PROJECT_NAME}_joint_controller_node PROPERTIES OUTPUT_NAME c1_joint_controller_node PREFIX "")
add_dependencies(${PROJECT_NAME}_joint_controller_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_joint_controller_node ${catkin_LIBRARIES})

add_executable(${PROJECT_NAME}_pose_controller_node src/c2_pose_controller_node.cpp)
set_target_properties(${PROJECT_NAME}_pose_controller_node PROPERTIES OUTPUT_NAME c2_pose_controller_node PREFIX "")
add_dependencies(${PROJECT_NAME}_pose_controller_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(${PROJECT_NAME}_pose_controller_node ${catkin_LIBRARIES})
[... 79 lines cut ...]
[8/18] Building CXX object CMakeFiles\abb_libegm_samples_joint_static_goal_node.dir\src\b1_joint_static_goal_node.cpp.obj
FAILED: CMakeFiles/abb_libegm_samples_joint_static_goal_node.dir/src/b1_joint_static_goal_node.cpp.obj
C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\cl.exe  /nologo /TP -DNOMINMAX -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"abb_libegm_samples\" -DWIN32_LEAN_AND_MEAN -D_USE_MATH_DEFINES -IC:\opt\ros\noetic\x64\include -IC:\opt\ros\noetic\x64\share\xmlrpcpp\cmake\..\..\..\include\xmlrpcpp /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /Zi /O2 /Ob1 /DNDEBUG   /D _VARIADIC_MAX=10 /Zc:__cplusplus /showIncludes /FoCMakeFiles\abb_libegm_samples_joint_static_goal_node.dir\src\b1_joint_static_goal_node.cpp.obj /FdCMakeFiles\abb_libegm_samples_joint_static_goal_node.dir\ /FS -c C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples\src\b1_joint_static_goal_node.cpp
C:\opt\ros\noetic\x64\include\boost/bind.hpp(41): note: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples\src\b1_joint_static_goal_node.cpp(38): fatal error C1083: Cannot open include file: 'abb_libegm/egm_trajectory_interface.h': No such file or directory
[9/18] Building CXX object CMakeFiles\abb_libegm_samples_joint_velocity_controller_node.dir\src\c3_joint_velocity_controller_node.cpp.obj
FAILED: CMakeFiles/abb_libegm_samples_joint_velocity_controller_node.dir/src/c3_joint_velocity_controller_node.cpp.obj
C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1428~1.293\bin\Hostx64\x64\cl.exe  /nologo /TP -DNOMINMAX -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"abb_libegm_samples\" -DWIN32_LEAN_AND_MEAN -D_USE_MAT [... cut ...]
C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples\src\c1_joint_controller_node.cpp(38): fatal error C1083: Cannot open include file: 'abb_libegm/egm_controller_interface.h': No such file or directory
ninja: build stopped: subcommand failed.
<== Failed to process package 'abb_libegm_samples':
  Command '['C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm\\env.bat', 'ninja', '-j8', '-l8']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples' && 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat' ninja -j8 -l8

Command failed, exiting.
'''
</details>

gavanderhoorn, 2021-03-15:
> I read [here](https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-692121293) the `abb_libegm_samples` were made some time ago when `abb_libegm` was a catkin package.

that's indeed the cause of the problem you encountered.

> I have updated my CMakeList.txt accordingly (I think, please correct me if I've done something wrong), here it is:

almost correct. The change to add a separate `find_package(abb_libegm ..)` is good, but:

> The new error I get when running `catkin_make_isolated` is the following:

your new error is caused by the fact you're missing: `target_link_libraries(... abb_libegm::abb_libegm ...)` for all targets built.

This will update the include paths to also include the location of the header files of `abb_libegm`, which is the cause of the error you show.

It will also cause `abb_libegm` to be linked to the targets, which would've probably been the next problem.

gavanderhoorn, 2021-03-15:
Note: this is actually a duplicate of the issue you already found.

The solution is mentioned in https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-740013963, which I believe you'd already also found.

As such, I'm going to close this issue.

Feel free to keep commenting on it.

closed 2021-03-15 by gavanderhoorn

DyumanAditya, 2021-03-15:
Thanks for your response,[ #18 (comment)](https://github.com/ros-industrial/abb_libegm/issues/18#issuecomment-740013963) didn't seem to be an explicit solution (to me at least): what do I need to add in place of the "..."  in `target_link_libraries(... abb_libegm::abb_libegm ...)`? 

Adding this:
```cmake
target_link_libraries(${PROJECT_NAME} PRIVATE
  abb_libegm::abb_libegm
  ${catkin_LIBRARIES}
)
```
gives me the following error:

<details><summary>Click to expand</summary>

```
==> Processing catkin package: 'abb_libegm_samples'
==> Creating build directory: 'build_isolated\abb_libegm_samples'
==> Building with env: 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat'
==> cmake C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples -DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples -DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja in 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples'
-- The C compiler identification is MSVC 19.28.29337.0
-- The CXX compiler identification is MSVC 19.28.29337.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm_samples
-- Using CMAKE_PREFIX_PATH: C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:/Users/Dyuman/catkin_ws1/devel_isolated/abb_libegm;C:\opt\ros\noetic\x64\tools\vcpkg\installed\x64-windows;c:\opt\ros\noetic\x64
-- This workspace overlays: c:/opt/ros/noetic/x64
-- Found PythonInterp: C:/opt/ros/noetic/x64/python.exe (found suitable version "3.8.3", minimum required is "3")
-- Using PYTHON_EXECUTABLE: C:/opt/ros/noetic/x64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\ros\noetic\x64\lib\site-packages\em.py
-- Using empy: C:/opt/ros/noetic/x64/lib/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/ros/noetic/x64/Scripts/nosetests.exe
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- Found Threads: TRUE
-- Found Boost: C:/opt/ros/noetic/x64/include (found version "1.73.0") found components: regex system thread chrono date_time atomic
CMake Error at CMakeLists.txt:23 (target_link_libraries):
  Cannot specify link libraries for target "abb_libegm_samples" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Dyuman/catkin_ws1/build_isolated/abb_libegm_samples/CMakeFiles/CMakeError.log".
<== Failed to process package 'abb_libegm_samples':
  Command '['C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm\\env.bat', 'cmake', 'C:\\Users\\Dyuman\\catkin_ws1\\src\\abb_libegm_samples', '-DCATKIN_DEVEL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\devel_isolated\\abb_libegm_samples', '-DCMAKE_INSTALL_PREFIX=C:\\Users\\Dyuman\\catkin_ws1\\install_isolated', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd 'C:\Users\Dyuman\catkin_ws1\build_isolated\abb_libegm_samples' && 'C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm\env.bat' cmake 'C:\Users\Dyuman\catkin_ws1\src\abb_libegm_samples' '-DCATKIN_DEVEL_PREFIX=C:\Users\Dyuman\catkin_ws1\devel_isolated\abb_libegm_samples' '-DCMAKE_INSTALL_PREFIX=C:\Users\Dyuman\catkin_ws1\install_isolated' -DCMAKE_BUILD_TYPE=RelWithDebInfo -G Ninja

Command failed, exiting.
```
</details>

I'm hoping to correct the errors and re-upload the `abb_libegm_samples` package so that others can build it successfully too.

gavanderhoorn, 2021-03-15:
As I wrote in https://github.com/ros-industrial/abb_libegm/issues/121#issuecomment-799491701:

> your new error is caused by the fact you're missing: `target_link_libraries(... abb_libegm::abb_libegm ...)` **for all targets built.**

All targets built by the `CMakeLists.txt` need to have that line added.

You could also add `abb_libegm::abb_libegm` to the `target_link_libraries(..)` lines already present for each binary target.

DyumanAditya, 2021-03-15:
Thank you for your help @gavanderhoorn! I got the package to build correctly. I'm uploading the updated `abb_libegm_samples` that will now build successfully on Windows (probably linux too)
[abb_libegm_samples.zip](https://github.com/ros-industrial/abb_libegm/files/6142974/abb_libegm_samples.zip)