ros-industrial/abb_libegm #123: Setting up EGM sessions for abb_libegm_samples

DyumanAditya, 2021-03-18:
Hi,
I apologize for my elementary questions.
My catkin_ws looks like this:
```
-- src
    -- abb_egm_rws_managers
    -- abb_libegm
    -- abb_libegm_samples
    -- abb_librws
    -- abb_robot_cpp_utilities
    -- abb_robot_driver_interfaces
    -- abb_rws_service_provider
```

What I am trying to do is to start an EGM session remotely and then use `rosrun` to launch a node from the `abb_libegm_samples`.
I have loaded the EGMSamplesSolution in RobotStudio meant for the `abb_libegm_samples`. 
When I run `roslaunch abb_rws_service_provider rws_service_provider.launch robot_ip:=127.0.0.1` to start the RWS services and then in a different terminal run `rosservice call /rws/pp_to_main "{}"` or `rosservice call /rws/start_rapid "{}"`, I get:
```
result_code: 3001
message: "\"Robot controller is not in AUTO mode\""
```

But in RobotStudio the controller operating mode is set to "auto". How can I fix this?

On a separate note, how can I start an EGM session from RobotStudio itself, without RWS? I've tried setting the Program Pointers to main and then running the rapid code from inside RobotStudio but that doesn't seem to do it.

jontje, 2021-03-18:
The `abb_rws_service_provider` node requires the `abb_rws_state_publisher` node for most services, as mentioned in the [overveiw](https://github.com/ros-industrial/abb_robot_driver/tree/master/abb_rws_service_provider#overview) section.

gavanderhoorn, 2021-03-18:
@jontje: tracking this in https://github.com/ros-industrial/abb_robot_driver/issues/22.

DyumanAditya, 2021-03-18:
> The abb_rws_service_provider node requires the abb_rws_state_publisher node for most services, as mentioned in the overveiw section.

Thank you, this solved my problem

gavanderhoorn, 2021-05-12:
Closing as it seems resolved.

closed 2021-05-12 by gavanderhoorn