ros-industrial/abb_libegm #126: What does the duration really mean?

ScottYang1982, 2021-04-08:
```
 p_point = trajectory_1.add_points();
 p_point->set_reach(true);
 p_point->set_duration(2.0);
```
Hi, here the duration, it's the time from start or just the time interval between current point to the next point?

Thanks! @jontje

jontje, 2021-04-09:
Hi @ScottYang1982,

That duration is the desired time between the current and next point, and it is used in the [egm_interpolator.cpp](https://github.com/ros-industrial/abb_libegm/blob/e0b8c3b0a906a50a7ad8d1d1b465a4343369c4e8/src/egm_interpolator.cpp) file.

Please note that there is no guarantee that the robot will be able to adhere to the desired duration. For example, that depends on how EGM have been configured on the robot controller side.

gavanderhoorn, 2021-05-12:
Assuming the question has been answered, I'm closing this.

closed 2021-05-12 by gavanderhoorn