ros-industrial/abb_libegm #150: Problems with compiling and using the abb_libegm library on Windows

parteimeister, 2022-12-19:
Are there any plans to add a manual on how to compile this library on Windows? It is an easy task for Linux, while it is absolute hell for Windows. After a week of investigation, I managed to assemble this library using additional strings in CMakeLists.txt, and I am sure that this is not an optimal solution:
```cmake
set(Boost_INCLUDE_DIR "C:/Users/username/source/libs/boost/boost-1.81.0")
set(Protobuf_LIBRARIES "C:/Users/username/source/libs/protobuf/protobuf-3.21.12/lib")
set(Protobuf_INCLUDE_DIR "C:/Users/username/source/libs/protobuf/protobuf-3.21.12/include")
set(Protobuf_PROTOC_EXECUTABLE "C:/username/User/source/libs/protobuf/protobuf-3.21.12/bin/protoc.exe")
option(BUILD_SHARED_LIBS ON)
```

The problem, for now, is that I need to figure out how to add compiled library to the new project correctly. I tried to add this strings in CMakeLists.txt of my project:
```cmake
set(CMAKE_PREFIX_PATH "C:/Users/username/source/libs/abb_libegm/share/abb_libegm ")
set(Boost_INCLUDE_DIR "C:/Users/username/source/libs/boost/boost-1.81.0")
find_package (abb_libegm REQUIRED)
target_link_libraries(abb_libegm_my_examples PUBLIC abb_libegm::abb_libegm)
```

After that, I have problems with linkage, e.g
```
error LNK2019: unresolved external symbol "public: virtual __cdecl abb::egm::wrapper::Header::~Header(void)"
```

Could someone help me with compiling this library on Windows?

traversaro, 2022-12-20:
Are you compiling your program an `abb_libegm` with the same build type (i.e. Release/Release or Debug/Debug)?

parteimeister, 2022-12-20:
@traversaro Yes. At the very beginning, I met with compilation problems because when installing the debug and release versions in the same folder, the files overwrite each other, so it is necessary to install debug and release versions of libraries in different folders. By the way, yesterday I successfully launched examples of using this library. The problem was that specifying ```Protobuf_LIBRARIES``` was not enough. It is necessary to manually specify paths to each library (dunno, this a protobuf-3.21.12 library bug or not). Anyway, this library can be compiled under Windows without ROS, I can close the issue.

By the way, I have an additional question. Can this library work with custom coordinate systems and different tools? And is it possible to get the current coordinates of the robot using this library? I did not find any interface for that.

closed 2022-12-20 by parteimeister